gazebosim/ros_gz #366: Memory Leak using ros_gz_bridge to convert LaserScan

retinfai, 2023-02-28:
## Environment
* OS Version: Ubuntu 22.04 LTS
* Source or binary build?
Source Build: `ros2` branch, `commit e7660f8`

## Description
### Expected behavior:
Potential Memory Leak

### Actual behavior:
Adding Lidar to the `ros_gz_bridge`, results in `gz sim server` consuming memory, going from 2% memory usage to 28.4% memory usage after running for over 10 minutes (this is repeatable)

## Steps to reproduce

1. `gz sim gpu_lidar_sensor.sdf`
2. `ros2 run ros_gz_bridge parameter_bridge /lidar@sensor_msgs/msg/LaserScan@gz.msgs.LaserScan`
3. Let it run for over 10 Minutes

## Output
**Note:** process `3978322` is the `gz sim server` process
[image]

For comparison, here is the same simulation after just restarting.
[image]

retinfai, 2023-03-01:
After more than an hour of running:
[image]

mjcarroll, 2023-03-01:
I think this is likely unrelated to the bridge itself.  The two ruby instances are going to be the gazebo server and gazebo gui.  To help track this down, can you verify which of the two that it is, and we can get it moved to the appropriate issue tracker.

retinfai, 2023-03-01:
Sweet, the process that's eating up the memory  - `process 10432` in the last image posted - is `gz sim server`. I'll re run the test without the bridge to confirm.

retinfai, 2023-03-01:
Another Update:
Left running for several hours with no memory leak, no `param` bridge:
[image]

However, after **turning on** visualise lidar, the memory problem stated before has arisen over **minutes**
[image]

retinfai, 2023-03-01:
To redefine the problem more specifically:

Memory problems (possible leakage) occur when **visualising the lidar** *or* there is a `parameter_bridge` converting `gz.msg.LaserScan` to `sensor_msgs/msgs/LaserScan`. 

*Note:* I retested with **no** visualise lidar but **with** a param bridge - this lead to memory leakage

mjcarroll, 2023-03-01:
I think it's likely that the Lidar doesn't do anything unless there are active subscribers.  Either the GUI or parameter_bridge will trigger it to start processing.  I'm going to transfer this issue to gz-sensors, as that is likely where the issue resides.

iche033, 2023-03-01:
Is this in Garden? We fixed a lidar memory leak issue in gz-rendering7 (Garden) not long ago. Maybe we just need to do another gz-rendering7 release: https://github.com/gazebosim/gz-rendering/pull/799

retinfai, 2023-03-01:
Yes we are using Garden. Ahh yep, that's exactly the issue thanks. I'm going to close this issue.

To receive the update, how do I go about doing this. Do I just redo the binary install, or do I source install?

closed 2023-03-01 by retinfai

iche033, 2023-03-01:
we'll need to make an gz-rendering release on our end. Once that's out, you should only just need to do a binary install to get the latest version.

iche033, 2023-03-04:
gz-rendering 7.3.0 has been released.

paoloelle, 2024-12-16:
@iche033 do you know if this problem still exists in Gazebo Fortress?

iche033, 2024-12-16:
I don't think so. The mem leak was from code added in Garden.