gazebosim/ros_gz #367: Depth camera intrinsics do not update

Veng97, 2023-03-06:
I'm using the depth camera plugin and describe my camera with the following SDF:

`    <gazebo reference="${name}_left_camera_frame">
      <sensor name="depth_camera" type="depth_camera">
        <topic>${name}/depth/image_rect</topic>
        <ignition_frame_id>${name}_left_camera_optical_frame</ignition_frame_id>
        <update_rate>${camera_update_rate}</update_rate>
        <visualize>true</visualize>
        <always_on>1</always_on>
        <camera name="${name}_left_camera_depth">
          
          
          <image>
            
            
            <format>R_FLOAT32</format>
          </image>
          <clip>
            <near>${minimum_depth_m}</near>
            <far>${maximum_depth_m}</far>
          </clip>
          <noise>
		<type>gaussian</type>
		<mean>0</mean>
		<stddev>0.01</stddev>
		</noise>
        </camera>
      </sensor>
    </gazebo>
`

I noticed that the camera info topic doesn't update when i override the default image width and height properties - that means that the camera intrinsics are incorrect when i specify non-default width and height parameters.
Note that i set the image frame by using the <ignition_frame_id> property.

azeey, 2023-03-07:
Hi @Veng97 , this looks like a bug request. Would you mind providing information requested in the [Bug Report template](https://github.com/gazebosim/ros_gz/issues/new?assignees=&labels=bug&template=bug_report.md)?

iche033, 2023-03-07:
There's been some work done on this recently: https://github.com/gazebosim/gz-sensors/pull/308. Maybe worth checking if you have the latest gz-sensors / ignition-sensors version

agonzat, 2023-05-31:
I can confirm that this bug is still present using the latest binary packages for ROS Humble on Ubuntu 22.04.

I tried to build a minimal example based on `ros2 launch ros_ign_gazebo_demos camera.launch.py` but these demos seem to be currently broken.

In our case, the intended sensor configuration was the following, which will produce CameraInfo messages that don't match the sensor parameters:

```xml
<sensor name="cameradepth" type="depth_camera">
  <topic>/cameradepth/depth_image</topic>
  <ignition_frame_id>camera_depth_optical_frame</ignition_frame_id>
  <camera name="camera">
    <horizontal_fov>1.57</horizontal_fov>
    <image>
      <width>1280</width>
      <height>720</height>
    </image>
    <clip>
      <near>0.1</near>
      <far>100</far>
    </clip>
    <noise>
      <type>gaussian</type>
      <mean>0.0</mean>
      <stddev>0.100</stddev>
    </noise>
  </camera>
  <always_on>1</always_on>
  <update_rate>30</update_rate>
  <visualize>0</visualize>
</sensor>
```
I obtain the following CameraInfo regardless of the configuration (`width`, `height` or `horizontal_fov`):

```yaml
header:
  stamp:
    sec: 11
    nanosec: 220000000
  frame_id: L_camera_depth_optical_frame
height: 240
width: 320
distortion_model: plumb_bob
d:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
k:
- 277.0
- 0.0
- 160.0
- 0.0
- 277.0
- 120.0
- 0.0
- 0.0
- 1.0
r:
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
p:
- 277.0
- 0.0
- 160.0
- -0.0
- 0.0
- 277.0
- 120.0
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
binning_x: 0
binning_y: 0
roi:
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: false
```

While removing the following lines makes the obtained CameraInfo messages match the Image:

```xml
<horizontal_fov>1.57</horizontal_fov>
<image>
    <width>1280</width>
    <height>720</height>
</image>
```

I experience the same issues on RGB cameras.