gazebosim/ros_gz #377: Actuator message support in ros_gz_bridge

gragleas, 2023-03-27:
## Desired behavior
The ability to use ros_gz_bridge parameter bridge for messages of the type gz.msgs.Actuator

## Alternatives considered
Send all messages for motor control via gz-transport instead of across ros_gz_bridge via a ros node.

## Additional context
It seems as though this was a work in progress some time ago according to [this](https://index.ros.org/p/ros_gz_bridge/) link. It was marked as TODO but the tag has since been removed from the official documentation. It looks like things on the gz side (formerly ignition) had some work done on them at one point, but was removed in #93. I was wondering if there has been any progress on this or if there is a possible workaround.

azeey, 2023-03-30:
/cc @bperseghetti

bperseghetti, 2023-03-30:
@gragleas we are adding Actuator messages upstream to ros_gz and also have updated multiple plugins AckermannSteering, JointController, and JointPositionController to also listen to an actuator message much like the existing MulticopterMotorModel Plugin. I'll share the links to the PRs when I have them up later today.

bperseghetti, 2023-04-01:
@azeey @gragleas Still need to stop being lazy and actually write the tests for a variety of the ones linked in this: https://github.com/gazebosim/ros_gz/pull/378

But just so it can be kept track of I included it here.

Also I need to create/join a release group for: https://github.com/rudislabs/actuator_msgs

azeey, 2023-05-26:
Fixed by https://github.com/gazebosim/ros_gz/pull/378 and #394.

closed 2023-05-26 by azeey