gazebosim/ros_gz #382: Add mappings for gz.msgs.AirSpeedSensor and gz.msgs.Altimeter

srmainwaring, 2023-04-19:
## Desired behavior

Provide mappings for sensor data:

- `gz.msgs.AirSpeedSensor`
- `gz.msgs.Altimeter`

## Alternatives considered

- NA.

## Implementation suggestion

The main issue is which ROS message type to target as these messages do not have a corresponding message in `sensor_msgs`.

## Additional context

Would be happy to submit a PR for this feature request once a suitable ROS message type is agree upon.

azeey, 2023-05-01:
We've been creating our own ROS messages when a compatible message type is not available in the ROS ecosystem (see https://github.com/gazebosim/ros_gz/tree/ros2/ros_gz_interfaces). Can we do the same here?

ahcorde, 2023-05-08:
Feel free to add me as a reviewer @srmainwaring

adityapande-1995, 2023-06-14:
SensorNoise msg work in progress : https://github.com/gazebosim/ros_gz/pull/417

wittenator, 2023-11-13:
Just for the sake of documentation: It seems that the Air Speed sensor msgs only exists in gz_msgs 9+. Since the current stable version of ros_gz is using Fortress and therefore gz_msgs version 7, there is no message to bridge atm.