gazebosim/ros_gz #383: Small error in the Warnings when creating a bridge with ros2

Mathosmls, 2023-04-28:
## Environment
* OS Version:  Ubuntu 22.04 
 * Source
* ros2 humble
* gazebo fortress

## Description
* Expected behavior: should print the ros2 type and the gazebo topic in the warning message
* Actual behavior: swap the ros2 type and the gazebo topic in the warning message

## Steps to reproduce
- launch a ros_gz_bridge with a full set of configuration as parameters with an error in it (I wrote std_msgs/msg/float64 instead of std_msgs/msg/Float64 for example)

my yaml : 

```
- ros_topic_name: "ros2gazebo/motor_back_left"
  ros_type_name: "std_msgs/msg/float64"
  gz_topic_name: "/model/world/joint/back_left_propeller_joint/cmd_pos"
  gz_type_name: "ignition.msgs.Double"
  direction:
    ROS_TO_GZ # Default "BIDIRECTIONAL" - Bridge both directions
    # "GZ_TO_ROS" - Bridge Ignition topic to ROS
    # "ROS_TO_GZ" - Bridge ROS topic to Ignition
```
my cmd : 
`ros2 run ros_gz_bridge parameter_bridge --ros-args -p config_file:=odrive_ros2gazebo/gazebo_bridge.yaml`

## Output
```
[INFO] [1682673937.724981799] [ros_gz_bridge]: Creating ROS->GZ Bridge: [ros2gazebo//motor_back_left (std_msgs/msg/float64) -> /model/world/joint/back_left_propeller_joint/cmd_pos (ignition.msgs.Double)] (Lazy 0)
[WARN] [1682673937.725289181] [ros_gz_bridge]: Failed to create a bridge for topic [ros2gazebo/motor_back_left] with ROS2 type [/model/world/joint/back_left_propeller_joint/cmd_pos] to topic [std_msgs/msg/float64] with Gazebo Transport type [ignition.msgs.Double]

```

azeey, 2023-04-28:
Thanks for reporting this. Would you be able to submit a PR to fix it?

Bi0T1N, 2023-04-28:
It's already fixed in the `ros2` branch by PR #371. Just needs to be cherry-picked into `humble` and `foxy` branch.

azeey, 2023-04-28:
Thanks @Bi0T1N! I opened #384 for humble.

Mathosmls, 2023-04-30:
Thanks guys !

PR #384 (2023-05-01): Backport: Fix warning message (#371)
# Backport

Fixes #383  

**Note to maintainers**: Remember to **Rebase  Merge**
closed 2023-04-30 by Mathosmls