gazebosim/ros_gz #385: Service bridging for ignition services that are not `WorldControl`?

jwdinius, 2023-04-30:
## Desired behavior

#211 added support for bridging services that have a request type of `WorldControl`.  I would like to be able to bridge `SetEntityPose` as well.

## Alternatives considered

Calling the ignition service in an OS process launched within a ROS 2 python node (using `subprocess`  or `os` package)

## Implementation suggestion

Could we not follow the pattern from #211?  I'm curious why support for other services was not added at this time, or why an explicit error message was not added when trying to bridge a service that did not have a `WorldControl` request type.  The current implementation keeps blocking even though nothing useful is being done when trying to bridge different service types.

## Additional context

azeey, 2023-05-01:
Looks like a duplicate of https://github.com/gazebosim/ros_gz/issues/363. There's a draft PR mapping `EntityPose` in https://github.com/gazebosim/ros_gz/pull/380. I'll go ahead and close this, but feel free to reopen if this is different from #363.

closed 2023-05-01 by azeey

MertCecen, 2024-03-29:
Is there any updates on this I need to use SetEntityPose but it's just not working.