gazebosim/ros_gz #389: Failed to create bridge for geometry_msgs/msg/Pose.

Rakshit-bit, 2023-05-09:
## Environment
* OS Version: Ubuntu 22.04 
* Source or binary build?
  I previously installed binary and it is still there. Then I installed the source version. For this I have also installed gazebo garden from source as I am  having gazebo 11 installed with ros2 humble.

## Description
* Expected behaviour: 
* Actual behaviour: 

## Steps to reproduce

1. Run any robot in gazebo. 
2. Do gz topic -l and get the pose topic and message
3. Then start the bridge: ros2 run ros_gz_bridge parameter_bridge 
4.

## Output

azeey, 2023-05-10:
Can you fill out the expected/actual behaviors and the output you get on the terminal? Also, using `ros_gz_bridge` with Garden and ROS Humble required building `ros_gz` with `GZ_VERSION=garden`. Can you verify that you've set the `GZ_VERSION` environment variable?

azeey, 2023-05-26:
Closing because the issue is incomplete and there hasn't been any activity.

closed 2023-05-26 by azeey