gazebosim/ros_gz #392: Gazebo topic remap

fbenti, 2023-05-20:
Hello, I was wondering if it is possible to remap a gazebo topic to a different name. 

So basically, I'm spawning a model with a camera, which publish on a topic `/camera`. Is it possible to remap it to `/camera_1` at runtime or it could be done only beforehand?

azeey, 2023-05-22:
You can remap topics using ROS once you've created a bridge, but topic remapping is not currently supported directly on Gazebo (gz-transport) topics. If you are requesting that as a feature, the issue should go into the gz-transport repo.

closed 2023-05-22 by azeey

caguero, 2023-05-22:
Gazebo topics can only be remapped programmatically. See this tutorial https://gazebosim.org/api/transport/12/messages.html in case that's useful. Otherwise, do it at the ROS level as @azeey recommended.