gazebosim/ros_gz #397: Build of `ros_gz_bridge` fails after PR #394

M1chaelM, 2023-05-23:
## Environment
* OS Version: 22.04
* Source or binary build?
source, humble branch. Last commits:
```
commit b64df31d2611ba4c51851f7270c3fb3096f94e9c (HEAD -> humble, tag: 0.244.11, origin/humble)
Author: Michael Carroll <michael@openrobotics.org>
Date:   Tue May 23 12:33:55 2023 -0500

    0.244.11
    
    Signed-off-by: Michael Carroll <michael@openrobotics.org>

commit 846323f299ee88ad6a01eb9ec140e10eb228169e
Author: Benjamin Perseghetti <bperseghetti@rudislabs.com>
Date:   Mon May 22 20:31:05 2023 -0400

    Add actuator_msgs to humble bridge. (#394)
    
    Signed-off-by: Benjamin Perseghetti <bperseghetti@rudislabs.com>

```

## Description
* Expected behavior: Running `colcon build` in the workspace should build all packages.
* Actual behavior: The build fails with an error on `ros_gz_bridge`

## Steps to reproduce
1. Clone `ros_gz` into a fresh workspace.
2. Run `GZ_VERSION=garden colcon build --merge-install`

## Output
```
--- stderr: ros_gz_bridge
CMake Error at CMakeLists.txt:81 (find_package):
  By not providing "Findactuator_msgs.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "actuator_msgs", but CMake did not find one.

  Could not find a package configuration file provided by "actuator_msgs"
  with any of the following names:

    actuator_msgsConfig.cmake
    actuator_msgs-config.cmake

  Add the installation prefix of "actuator_msgs" to CMAKE_PREFIX_PATH or set
  "actuator_msgs_DIR" to a directory containing one of the above files.  If
  "actuator_msgs" provides a separate development package or SDK, be sure it
  has been installed.

gmake: *** [Makefile:1955: cmake_check_build_system] Error 1
```
## Note
I think this is related to PR #394 because the change happened very recently. I noticed it because it broke our CI (https://github.com/osrf/vrx/pull/658), so it seems to affect a fresh install.

mjcarroll, 2023-05-23:
Yes, that PR introduced a new dependency on `actuator_msgs`.  If you are installing via binary debians, this will be taken care of automatically.

If you are installing from source, you can get the dependency by installing `ros-humble-actuator-msgs`.

For robustness, I would recommend having a `rosdep install --from-paths src` as part of your source CI/installation process, which can catch added dependencies in `package.xml` for you.

M1chaelM, 2023-05-23:
Thanks for the speedy reply! Since this wasn't a bug I'll close it.

closed 2023-05-23 by M1chaelM

M1chaelM, 2023-05-27:
@mjcarroll Regarding your note about installing from binary debians--I would love to move to this but I thought they were not available for humble/garden. Is that correct?

caguero, 2023-05-27:
See this issue for instructions about how to install debs for Humble/Garden:

https://github.com/gazebo-tooling/release-tools/issues/895