gazebosim/ros_gz #401: Packages/stacks could not have their rosdep keys resolved to system dependencies (Ubuntu20.04 in WSL, ROS Humble, Ign Fortress)

Vincenzo-Saffioti, 2023-05-26:
## Environment
* OS Version: Ubuntu 20.04 Focal in WSL
* ROS2 Humble + Ignition Fortress
* Source build

## Description
I cannot install ros_gz either via binary or source. I need to use Ubuntu 20.04 and install Humble+Ignition Fortress. I installed Humble via source and used the Fortress binary.
However, I encountered difficulties when installing the bridge.
In particular, trying to install via source the humble branch packages I encourred problems in managing dependencies with rosdep:

```
rosdep install -r --from-paths src -i -y --rosdistro humble

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ros_gz_sim_demos: No definition of [xacro] for OS version [focal]
ros_gz_bridge: No definition of [actuator_msgs] for OS version [focal]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```

Clearly trying to build the packages I consequently get the following error in relation to the lack of actuator messages:
```
colcon build

Starting >>> ros_gz_interfaces
Starting >>> ros_gz_sim
[Processing: ros_gz_interfaces, ros_gz_sim]
Finished <<< ros_gz_sim [52.0s]
Starting >>> ros_ign_gazebo
Finished <<< ros_gz_interfaces [52.5s]
Starting >>> ros_gz_bridge
Starting >>> ros_ign_interfaces
--- stderr: ros_gz_bridge
CMake Error at CMakeLists.txt:81 (find_package):
  By not providing "Findactuator_msgs.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "actuator_msgs", but CMake did not find one.

  Could not find a package configuration file provided by "actuator_msgs"
  with any of the following names:

    actuator_msgsConfig.cmake
    actuator_msgs-config.cmake

  Add the installation prefix of "actuator_msgs" to CMAKE_PREFIX_PATH or set
  "actuator_msgs_DIR" to a directory containing one of the above files.  If
  "actuator_msgs" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< ros_gz_bridge [17.6s, exited with code 1]
Aborted  <<< ros_ign_gazebo [18.3s]
Aborted  <<< ros_ign_interfaces [25.3s]

Summary: 2 packages finished [1min 18s]
  1 package failed: ros_gz_bridge
  2 packages aborted: ros_ign_gazebo ros_ign_interfaces
  1 package had stderr output: ros_gz_bridge
  7 packages not processed
```

## Steps to reproduce

1. Install Ubuntu 20.04 in WSL
2. Install ROS2 Humble via source 
3. Install Ignition Fortress
4. Try to build ros_gz package

ahcorde, 2023-05-26:
It's wierd, the dependency is listed in the `package.xml` in the `humble` branch https://github.com/gazebosim/ros_gz/blob/humble/ros_gz_bridge/package.xml#LL16C11-L16C24. 

anywho you can compile `actuator_msgs` from source

robertorovella91, 2023-05-26:
I have the same problem. What about instead for "ros_gz_sim_demos: No definition of [xacro] for OS version [focal]" ?

Vincenzo-Saffioti, 2023-05-26:
> It's wierd, the dependency is listed in the `package.xml` in the `humble` branch https://github.com/gazebosim/ros_gz/blob/humble/ros_gz_bridge/package.xml#LL16C11-L16C24.
> 
> anywho you can compile `actuator_msgs` from source

Thank you for your suggestion. I tried to add manually `actuator_msgs` (https://github.com/rudislabs/actuator_msgs) and `xacro` (https://github.com/ros/xacro/tree/ros2) packages to my workspace but i had other problems that i don't understand, here's the log:

```
colcon build

Starting >>> ros_gz_interfaces
Starting >>> actuator_msgs
Starting >>> ros_gz_sim
Starting >>> xacro
Finished <<< ros_gz_sim [0.89s]
Starting >>> ros_ign_gazebo
Finished <<< ros_ign_gazebo [0.56s]
Finished <<< ros_gz_interfaces [8.26s]
Starting >>> ros_ign_interfaces
Finished <<< xacro [10.8s]
Finished <<< actuator_msgs [13.0s]
Starting >>> ros_gz_bridge
Finished <<< ros_ign_interfaces [22.0s]
[Processing: ros_gz_bridge]
[Processing: ros_gz_bridge]
[Processing: ros_gz_bridge]
[Processing: ros_gz_bridge]
[Processing: ros_gz_bridge]
[Processing: ros_gz_bridge]
[Processing: ros_gz_bridge]
[Processing: ros_gz_bridge]
[Processing: ros_gz_bridge]
--- stderr: ros_gz_bridge
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [CMakeFiles/ros_gz_bridge_lib.dir/build.make:441: CMakeFiles/ros_gz_bridge_lib.dir/generated/factories/geometry_msgs.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [CMakeFiles/ros_gz_bridge_lib.dir/build.make:480: CMakeFiles/ros_gz_bridge_lib.dir/generated/factories/ros_gz_interfaces.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [CMakeFiles/ros_gz_bridge_lib.dir/build.make:519: CMakeFiles/ros_gz_bridge_lib.dir/generated/factories/std_msgs.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [CMakeFiles/test_ros_subscriber.dir/build.make:151: CMakeFiles/test_ros_subscriber.dir/generated/test/geometry_msgs_subscriber.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:487: CMakeFiles/test_ros_subscriber.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:321: CMakeFiles/ros_gz_bridge_lib.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< ros_gz_bridge [4min 57s, exited with code 2]

Summary: 6 packages finished [5min 11s]
  1 package failed: ros_gz_bridge
  1 package had stderr output: ros_gz_bridge
  7 packages not processed
```

ahcorde, 2023-05-26:
something is wrong, probably with the amount of memory that you have in your system.

```bash
c++: fatal error: Killed signal terminated program cc1plus
```
If you are running this in VM increase the virtual memory or you can add this option to colcon `--parallel-workers 1`

azeey, 2023-05-30:
ROS Humble officially only supports Ubuntu 22.04 (https://docs.ros.org/en/humble/Installation.html). That's why the actuator_msgs package is missing. When building from source, I agree with @ahcorde, I think you're running out of memory.  Try adding `--parallel-workers=1 --executor sequential` to your `colcon` command to ensure only one instance of the compiler is running at a time.

Vincenzo-Saffioti, 2023-05-31:
> ROS Humble officially only supports Ubuntu 22.04 (https://docs.ros.org/en/humble/Installation.html). That's why the actuator_msgs package is missing. When building from source, I agree with [...]

Thank you @azeey @ahcorde for helping me, I was able to build all the packages correctly by increasing the amount of RAM. I will try to run some tests to make sure everything is working!

djsamseng, 2023-06-23:
For anyone else who might run into this issue:

 On Ubuntu 20.04 focal, ROS 2 Humble, Ignition Fortress when having the following error
```
--- stderr: ros_gz_bridge
CMake Error at CMakeLists.txt:81 (find_package):
  By not providing "Findactuator_msgs.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "actuator_msgs", but CMake did not find one.

  Could not find a package configuration file provided by "actuator_msgs"
  with any of the following names:

    actuator_msgsConfig.cmake
    actuator_msgs-config.cmake

  Add the installation prefix of "actuator_msgs" to CMAKE_PREFIX_PATH or set
  "actuator_msgs_DIR" to a directory containing one of the above files.  If
  "actuator_msgs" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< ros_gz_bridge [17.6s, exited with code 1]
```

Simply
```bash
cd src
git clone git@github.com:rudislabs/actuator_msgs.git
cd ../
colcon build
```

fixed the issue.

I was then able to confirm ros and ignition fortress working correctly by doing the following:

[following this guide with modifications for an Ubuntu 20.04 local installation of ROS2 Humble and gazebo fortress](https://gazebosim.org/docs/garden/ros2_integration)

```bash
cd ~/dev/gazebo_fortress_ws/
source ~/dev/ros2_humble/install/local_setup.bash
source ./install/local_setup.bash
ros2 run ros_gz_bridge parameter_bridge /keyboard/keypress@std_msgs/msg/Int32@gz.msgs.Int32
```
In a second terminal
```bash
ign gazebo sim empty.sdf
```
- Note replace the `gz` command with `ign gazebo` since the `gz` command was not yet introduced for fortress
- In the simulator window click the dropdown in the top right
and add `Key Publisher`

In a third terminal
```bash
source ~/dev/ros2_humble/install/local_setup.bash
ros2 topic echo /keyboard/keypress
```

Press buttons on the keyboard when focused in the simulator and see the keypressed logged in the third terminal

azeey, 2023-12-01:
Fixed by #431.

closed 2023-12-01 by azeey