gazebosim/ros_gz #402: Error when trying to build the ros_gz package from source

Mathosmls, 2023-05-30:
## Environment
Ubuntu 20.04 
ROS2 Humble, from binary
Gazebo Garden, from binary

## Description
* Expected behavior: the package should be build without errors
* Actual behavior: error when trying to build the ros_gz_bridge package

## Steps to reproduce

1. install ros2 humble from binary (instructions from this page https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html)
2. install gazebo garden from binary ( https://gazebosim.org/docs/garden/install_ubuntu#binary-installation-on-ubuntu)
3. follow the step here for an unsupported Gazebo version with ROS 2  (https://gazebosim.org/docs/garden/ros_installation)
4. Try to install ros_gz from source (instruction at the end of the readme)

## Output
```
Starting >>> ros_gz_interfaces
Starting >>> ros_gz_sim
Finished <<< ros_gz_sim [0.24s]                                           
Finished <<< ros_gz_interfaces [0.76s]                     
Starting >>> ros_gz_bridge
--- stderr: ros_gz_bridge                             
Traceback (most recent call last):
  File "/home/mathis/ws/src/ros_gz/ros_gz_bridge/bin/ros_gz_bridge_generate_factories", line 25, in <module>
    from ros_gz_bridge import generate_cpp
  File "/home/mathis/ws/src/ros_gz/ros_gz_bridge/ros_gz_bridge/__init__.py", line 21, in <module>
    from rosidl_pycommon import expand_template
ModuleNotFoundError: No module named 'rosidl_pycommon'
gmake[2]: *** [CMakeFiles/ros_gz_bridge_lib.dir/build.make:80: generated/get_factory.cpp] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:182: CMakeFiles/ros_gz_bridge_lib.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ros_gz_bridge [5.06s, exited with code 2]

Summary: 2 packages finished [6.00s]
  1 package failed: ros_gz_bridge
  1 package had stderr output: ros_gz_bridge
  3 packages not processed

```

Mathosmls, 2023-05-30:
Okay so I found the solution thanks to #300 , you have to clone the branch humble and not the ros2 one. Perhaps modifying the readme and/or the tutorial on gazebosim.org might be a good idea.

closed 2023-05-30 by Mathosmls

gorghino, 2023-05-31:
Hi @Mathosmls 
How did you install Humble with binaries on Ubuntu 20.04? ros-humble-desktop is available for Ubuntu 22.04

Mathosmls, 2023-05-31:
Hi @gorghino 
oops I forgot to change the default value ^^' I am on Ubuntu 22.04 off course. Thanks for noticing that !