gazebosim/ros_gz #409: Not able to Control Robot in gazebo using teleop_twsit_joy package

Rakshit-bit, 2023-06-10:
## Environment
* OS Version: 
* Source or binary build?
  **Build from source ros2 Humble branch**
  

## Description

- * Expected behavior:  Robot in gazebo garden should move using teleop_twist_joy package. Joystick is general (any) as the teleop_package suggest. -->
- * Actual behavior: I created my urdf file of the robot and added the ackermann_system plugin of gazebo garden to it.
- * Using the ros_gz_bridge, I am  passing my model's cmd_vel topic to gazebo (ROS->GZ) and seeing the topic using both ros2 topic list and gz topic -l.
- * I run my launch file of teleop_twist_joy node and conected my joystick (verified that it is detected). 
- * Now, when I press buttons on the joystick, and check in new terminal with command = ros2 topic echo /joy, it is publishing the messages.
- * But when I do ros2 topic echo /model/podcar/cmd_vel, it doesn't show anything there niether from gazebo side (gz topic -e --topic /model/podcar/cmd_vel).
- * To verify that the bridge is working fine, I killed the gazebo sim and i tested normally by creating a bridge of cmd_vel topic from ROS->GZ and ran the general teleop_twist_joy command (ros2 launch teleop_twist_joy teleop-launch.py joy) with default configurations.
- * This time, it is publishing on /cmd_vel and gazebo is receiving that /cmd_vel, I can see the messages in linear x and angular z.
- * Is this related to the gazebo garden sim then, that it is not receiving the messages?
- * One thing more, my launch file launches the bridge, the gazebi_sim, loads my urdf model.
- * The other launch file which launches teleop_twist_joy node is in different package, I source everything correctly and run it.
- * But do i have to provide the package dependency of the paackage containing my teleop_twsit_joy launch file to the package which is running the ros_gz_sim, the bridge? I dont think this is the issue here. 

## Steps to reproduce
**My first launch file**

1.```
import os
from launch import LaunchDescription
from launch_ros.actions import Node
from launch.actions import DeclareLaunchArgument, IncludeLaunchDescription, ExecuteProcess
from launch.substitutions import LaunchConfiguration, Command, PathJoinSubstitution
from launch_ros.substitutions import FindPackageShare
from launch.conditions import IfCondition
from ament_index_python.packages import get_package_share_path, get_package_share_directory
from launch_ros.parameter_descriptions import ParameterValue
from launch.launch_description_sources import PythonLaunchDescriptionSource

def generate_launch_description():
    
	world_path = os.path.join(get_package_share_directory('pod2_description'))
	path_to_urdf = PathJoinSubstitution([FindPackageShare("pod2_description"), "xacro", "tested.urdf"])
	
	rviz_config_path = PathJoinSubstitution([FindPackageShare("pod2_description"), "rviz2", "description.rviz"])
	
	robot_description = ParameterValue(
        Command(['xacro ', str(get_package_share_path('pod2_description') / 'xacro/tested.urdf')]),
        value_type=str)

	pkg_ros_gz_sim = get_package_share_directory('ros_gz_sim')
	gazebo_sim = IncludeLaunchDescription(
		PythonLaunchDescriptionSource(
		os.path.join(pkg_ros_gz_sim, 'launch', 'gz_sim.launch.py')
        ),
	launch_arguments={'gz_args': PathJoinSubstitution([
		world_path, 'xacro', 'model.sdf'
    ])}.items(),
    )
	spawn = Node(
		package='ros_gz_sim',
		executable='create',
		arguments=['-name', 'podcar',
	     '-topic', '/robot_description',
	    '-x', '0', '-y', '0', '-z', '0.5'],
	     output='screen'
    )
	# creating the (ROS-->GZ) bridge for podcar cmd-vel topic, (GZ-->ROS) for odometry topic, lidar data
	ros_gz_bridge = Node(
		package='ros_gz_bridge',
		name = 'ros_gz_bridge',
		executable='parameter_bridge',
		arguments=['/model/podcar/cmd_vel@geometry_msgs/msg/Twist]ignition.msgs.Twist',
	     '/model/podcar/odometry@nav_msgs/msg/Odometry[ignition.msgs.Odometry',
	 '/scan@sensor_msgs/msg/LaserScan[ignition.msgs.LaserScan',]
    )
    
	
	return LaunchDescription([
	DeclareLaunchArgument(
            name='urdf', 
            default_value=path_to_urdf,
            description='URDF path'
        ),
        
        # DeclareLaunchArgument(
        #     name='publish_joints', 
        #     default_value='true',
        #     description='Launch joint_states_publisher'
        # ),

        DeclareLaunchArgument(
            name='rviz', 
            default_value='false',
            description='Run rviz'
        ),

        DeclareLaunchArgument(
            name='use_sim_time', 
            default_value='false',
            description='Use simulation time'
        ),
        
    #     ExecuteProcess(
	# 	cmd=['ign', 'gazebo', '-r',
    #    os.path.join(world_path)]),
        # Node(
        #     package='joint_state_publisher',
        #     executable='joint_state_publisher',
        #     name='joint_state_publisher',
        #     condition=IfCondition(LaunchConfiguration("publish_joints"))
        #     # parameters=[
        #     #     {'use_sim_time': LaunchConfiguration('use_sim_time')}
        #     # ] #since galactic use_sim_time gets passed somewhere and rejects this when defined from launch file
        # ),

        Node(
            package='robot_state_publisher',
            executable='robot_state_publisher',
            name='robot_state_publisher',
            output='screen',
            parameters=[{'use_sim_time': LaunchConfiguration('use_sim_time'),
                    'robot_description': robot_description}]),      
            
        Node(
            package='rviz2',
            executable='rviz2',
            name='rviz2',
            output='screen',
            arguments=['-d', rviz_config_path],
            condition=IfCondition(LaunchConfiguration("rviz")),
            parameters=[{'use_sim_time': LaunchConfiguration('use_sim_time')}]
        ),
        gazebo_sim,
	    spawn,
	    ros_gz_bridge,
	    
    ])
```

2. **My second launch file** 
```
import rclpy
from launch_ros.actions import Node
from launch.actions import DeclareLaunchArgument, ExecuteProcess, IncludeLaunchDescription
from launch.substitutions import LaunchConfiguration, Command, PathJoinSubstitution
from launch_ros.substitutions import FindPackageShare
from launch import LaunchDescription

def generate_launch_description():

    return LaunchDescription([
        Node(
        package='joy',
        executable='joy_node',
        name='joy_node',
        output='screen',
        ),

        Node(
            package='teleop_twist_joy',
            executable='teleop_node',
            name='teleop_twist_joy_node',
            output ='screen',
            remappings=[('/cmd_vel', '/model/podcar/cmd_vel')]
        )
    ])

```

## Output

-  **Also, when I just try the normal ros2 topic pub command, the robot is moving fine but I want to use joystick,  as explained everything above.**

azeey, 2023-06-22:
I see you're using ROS 2 Humble, with Garden. This requires building `ros_gz` from source with the environment variable `GZ_VERSION=garden` . Have you done that? I also see you're using `ignition.msgs.Twist`. With the Ignition->Gazebo transition, you cannot use `ignition` message types. You'll need to use `gz.msgs.Twist`.

> But when I do ros2 topic echo /model/podcar/cmd_vel, it doesn't show anything there niether from gazebo side (gz topic -e --topic /model/podcar/cmd_vel).

This seems odd. If the mapping from your joy node's `cmd_vel` to `/model/podcar/cmd_vel` is working properly, you should be able to see it when you echo that topic in ROS even if Gazebo is not running. Can you verify this mapping is working correctly?

Rakshit-bit, 2023-06-23:
Yes, I have build the ros_gz from source. Also, I am using ignition.mags. (ignition namespace) for other messages but its working. Thank you for the advice for using gz.msgs.
I figured out that clock message bridge should be created in order for ROS to publish message in sim time. Also, we have to set use_sim_true: 'true' in teleop_twist_joy_node launch file.

Update-->> I am now able to control the angular z message from the joystick but linear x (for forward movement) is still no happening but I know that is the case with the joystick config (.yaml) file. I have to look into the config. I am using logitech 3D pro, do you know what config to set for it?

azeey, 2023-06-23:
Glad things are working!

>  have to look into the config. I am using logitech 3D pro, do you know what config to set for it?

Sorry, I don't have that controller, so I don't know what the config would be. 

Since this doesn't seem to be a `ros_gz` bug, I'll go ahead and close the issue.

closed 2023-06-23 by azeey