gazebosim/ros_gz #410: tf Ros_humble gz bug.

chngdickson, 2023-06-10:
## Environment
* OS Version: 

## Description
* Expected behavior: 
* Actual behavior: 

## Steps to reproduce

1. Write launch description as below
2. try not to cry
3.
```
def generate_launch_description():
  tf_bridge = Node(
    package='ros_gz_bridge',
    executable='parameter_bridge',
    # topic@ros_type@gazebo_type
    arguments=[
        f'/world/iris_runway2/model/iris_ardu_camera/joint_state@sensor_msgs/msg/JointState@gz.msgs.Model',
        f'/world/iris_runway2/pose/info@tf2_msgs/msg/TFMessage@gz.msgs.Pose_V'],
    remappings=[
        (f'/world/iris_runway2/model/iris_ardu_camera/joint_state', '/joint_states'),
        (f'/world/iris_runway2/pose/info', '/tf')
        ],
    output='screen'
    )
  
  return LaunchDescription([
    tf_bridge
  ])
```

## Output

thibs-sigma, 2023-07-11:
Hi all, 
Any news on that issue? I have the same problem. 
Could you provide some hints to try to debug that and potentially fix the issue "by hand" for the moment?

chngdickson, 2023-07-22:
> Hi all, Any news on that issue? I have the same problem. Could you provide some hints to try to debug that and potentially fix the issue "by hand" for the moment?

Dont think they have solved it yet. As you can see, the issue is sent to a TODO in core development. We just kinda have to wait patiently.

The workaround would be have all the poses in a single file. Which kinda defeats the purpose of an include. Any include will cause the tf to not get published correctly.

azeey, 2024-01-04:
It's best to use the `PosePublisher` system for working with `tf` (eg. https://github.com/gazebosim/ros_gz/blob/ros2/ros_gz_sim_demos/launch/tf_bridge.launch.py). The `pose/info`  topic is published by the `SceneBroadcaster` system, which is meant to be used by the GUI.