gazebosim/ros_gz #440: robot_description not accessible from custom ros2 controller

fafux, 2023-07-21:
## Environment
* OS Version: Ubuntu 22.04 
* ROS2 version: humble
* Source or binary build?
  ros-humble-ros-gz/jammy,now 0.244.11-1jammy.20230624.051759 amd64

## Description
* Behavior: Launching gz_sim.launch.py and create to spawn the urdf, parsing the xacro in which ign_ros2_control::IgnitionROS2ControlPlugin is contained, I expect the robot_description to be a parameter of the controller_manager but it's not. I want to access from a custom ros2 controller to the robot_description. Since it's not a parameter of the controller manager, the only way to access it is from a client (both service or parameter). In both cases, waiting the response of the future blocks the execution.

azeey, 2024-01-04:
Should this be filed at https://github.com/ros-controls/gz_ros2_control?
In general, `gz_sim.launch.py` doesn't publish anything to `robot_description`. You'd have to do that from your launch file. [Here](https://github.com/gazebosim/ros_gz/blob/e348e3be89a871e3d8388e79682da173ff743dc3/ros_gz_sim_demos/launch/joint_states.launch.py#L32-L45) is a good example.

fafux, 2024-01-08:
The standard `ros2_control_node` publishes the `robot_description` parameter which is passed to the node in the launch file as for the `robot_state_publisher`. For gazebo this is not possible. However, if I try to get the `robot_description` parameter with a parameter client to the `robot_state_publisher` node inside my controller, the execution is blocked to wait the response of the future callback.