gazebosim/ros_gz #444: Auto-play argument

mjforan, 2023-08-16:
## Desired behavior
In the ROS 1 version of this package (ros_ign), the simulator would "play" as soon as it started running. In the current ROS 2 version I have to hit the play button every time. I can understand how some people would prefer this new behavior, but it is a pain for me. Could this be exposed as a launch argument? I would like to be able to choose if the simulator should start up right away or wait for user input.

mjcarroll, 2023-08-17:
The command line parameter exists for gz-sim, but it hasn't been exposed as part of the launch file.  It should be relatively straightforward to add here: https://github.com/gazebosim/ros_gz/blob/0882b17e41cee70f365536adc3b2da72662c683a/ros_gz_sim/launch/gz_sim.launch.py.in#L63-L70

The `-r` CLI flag will cause the simulation to start as soon as gazebo is loaded.

mjforan, 2023-08-18:
Thanks for the tip, adding `-r` to `gz_args` works.

closed 2023-08-18 by mjforan