gazebosim/ros_gz #454: Installing ros-iron-ros-gz installs Gazebo Garden instead of Fortress

wittenator, 2023-10-12:
## Environment
* OS Version: 22.04
* Source or binary build?
  Binary

## Description
* Expected behavior: Gazebo Fortress should be installed
* Actual behavior: Gazebo Garden gets installed

## Steps to reproduce

1. Execute `sudo apt-get install ros-iron-ros-gz`

## Output
```
sudo apt install ros-iron-ros-gz
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  gz-plugin2-cli gz-sim7-cli gz-tools2 gz-transport12-cli libgz-cmake3-dev libgz-common5 libgz-common5-av libgz-common5-av-dev libgz-common5-core-dev
  libgz-common5-dev libgz-common5-events libgz-common5-events-dev libgz-common5-geospatial libgz-common5-geospatial-dev libgz-common5-graphics
  libgz-common5-graphics-dev libgz-common5-io libgz-common5-io-dev libgz-common5-profiler libgz-common5-profiler-dev libgz-common5-testing libgz-common5-testing-dev
  libgz-fuel-tools8 libgz-fuel-tools8-dev libgz-gui7 libgz-gui7-dev libgz-math7 libgz-math7-dev libgz-math7-eigen3-dev libgz-msgs9 libgz-msgs9-dev libgz-physics6
  libgz-physics6-bullet libgz-physics6-bullet-dev libgz-physics6-core-dev libgz-physics6-dartsim libgz-physics6-dartsim-dev libgz-physics6-dev
  libgz-physics6-heightmap-dev libgz-physics6-mesh-dev libgz-physics6-sdf-dev libgz-physics6-tpe libgz-physics6-tpe-dev libgz-physics6-tpelib
  libgz-physics6-tpelib-dev libgz-plugin2 libgz-plugin2-dev libgz-rendering7 libgz-rendering7-core-dev libgz-rendering7-dev libgz-rendering7-ogre1
  libgz-rendering7-ogre1-dev libgz-rendering7-ogre2 libgz-rendering7-ogre2-dev libgz-sensors7 libgz-sensors7-air-pressure libgz-sensors7-air-pressure-dev
  libgz-sensors7-air-speed libgz-sensors7-air-speed-dev libgz-sensors7-altimeter libgz-sensors7-altimeter-dev libgz-sensors7-boundingbox-camera
  libgz-sensors7-boundingbox-camera-dev libgz-sensors7-camera libgz-sensors7-camera-dev libgz-sensors7-core-dev libgz-sensors7-depth-camera
  libgz-sensors7-depth-camera-dev libgz-sensors7-dev libgz-sensors7-force-torque libgz-sensors7-force-torque-dev libgz-sensors7-gpu-lidar
  libgz-sensors7-gpu-lidar-dev libgz-sensors7-imu libgz-sensors7-imu-dev libgz-sensors7-lidar libgz-sensors7-lidar-dev libgz-sensors7-logical-camera
  libgz-sensors7-logical-camera-dev libgz-sensors7-magnetometer libgz-sensors7-magnetometer-dev libgz-sensors7-navsat libgz-sensors7-navsat-dev
  libgz-sensors7-rendering libgz-sensors7-rendering-dev libgz-sensors7-rgbd-camera libgz-sensors7-rgbd-camera-dev libgz-sensors7-segmentation-camera
  libgz-sensors7-segmentation-camera-dev libgz-sensors7-thermal-camera libgz-sensors7-thermal-camera-dev libgz-sensors7-wide-angle-camera
  libgz-sensors7-wide-angle-camera-dev libgz-sim7 libgz-sim7-dev libgz-sim7-plugins libgz-tools2-dev libgz-transport12 libgz-transport12-core-dev
  libgz-transport12-dev libgz-transport12-log libgz-transport12-log-dev libgz-transport12-parameters libgz-transport12-parameters-dev libgz-utils2
  libgz-utils2-cli-dev libgz-utils2-dev libogre-next-2.3-dev libogre-next-2.3.0 libsdformat13 libsdformat13-dev ros-iron-actuator-msgs
  ros-iron-compressed-depth-image-transport ros-iron-compressed-image-transport ros-iron-image-transport-plugins ros-iron-ros-gzgarden-bridge
  ros-iron-ros-gzgarden-image ros-iron-ros-gzgarden-interfaces ros-iron-ros-gzgarden-sim ros-iron-ros-gzgarden-sim-demos ros-iron-sdformat-urdf
  ros-iron-theora-image-transport ros-iron-xacro sdformat13-sdf
Suggested packages:
  ogre-next-2.3-doc
The following NEW packages will be installed:
  gz-plugin2-cli gz-sim7-cli gz-tools2 gz-transport12-cli libgz-cmake3-dev libgz-common5 libgz-common5-av libgz-common5-av-dev libgz-common5-core-dev
  libgz-common5-dev libgz-common5-events libgz-common5-events-dev libgz-common5-geospatial libgz-common5-geospatial-dev libgz-common5-graphics
  libgz-common5-graphics-dev libgz-common5-io libgz-common5-io-dev libgz-common5-profiler libgz-common5-profiler-dev libgz-common5-testing libgz-common5-testing-dev
  libgz-fuel-tools8 libgz-fuel-tools8-dev libgz-gui7 libgz-gui7-dev libgz-math7 libgz-math7-dev libgz-math7-eigen3-dev libgz-msgs9 libgz-msgs9-dev libgz-physics6
  libgz-physics6-bullet libgz-physics6-bullet-dev libgz-physics6-core-dev libgz-physics6-dartsim libgz-physics6-dartsim-dev libgz-physics6-dev
  libgz-physics6-heightmap-dev libgz-physics6-mesh-dev libgz-physics6-sdf-dev libgz-physics6-tpe libgz-physics6-tpe-dev libgz-physics6-tpelib
  libgz-physics6-tpelib-dev libgz-plugin2 libgz-plugin2-dev libgz-rendering7 libgz-rendering7-core-dev libgz-rendering7-dev libgz-rendering7-ogre1
  libgz-rendering7-ogre1-dev libgz-rendering7-ogre2 libgz-rendering7-ogre2-dev libgz-sensors7 libgz-sensors7-air-pressure libgz-sensors7-air-pressure-dev
  libgz-sensors7-air-speed libgz-sensors7-air-speed-dev libgz-sensors7-altimeter libgz-sensors7-altimeter-dev libgz-sensors7-boundingbox-camera
  libgz-sensors7-boundingbox-camera-dev libgz-sensors7-camera libgz-sensors7-camera-dev libgz-sensors7-core-dev libgz-sensors7-depth-camera
  libgz-sensors7-depth-camera-dev libgz-sensors7-dev libgz-sensors7-force-torque libgz-sensors7-force-torque-dev libgz-sensors7-gpu-lidar
  libgz-sensors7-gpu-lidar-dev libgz-sensors7-imu libgz-sensors7-imu-dev libgz-sensors7-lidar libgz-sensors7-lidar-dev libgz-sensors7-logical-camera
  libgz-sensors7-logical-camera-dev libgz-sensors7-magnetometer libgz-sensors7-magnetometer-dev libgz-sensors7-navsat libgz-sensors7-navsat-dev
  libgz-sensors7-rendering libgz-sensors7-rendering-dev libgz-sensors7-rgbd-camera libgz-sensors7-rgbd-camera-dev libgz-sensors7-segmentation-camera
  libgz-sensors7-segmentation-camera-dev libgz-sensors7-thermal-camera libgz-sensors7-thermal-camera-dev libgz-sensors7-wide-angle-camera
  libgz-sensors7-wide-angle-camera-dev libgz-sim7 libgz-sim7-dev libgz-sim7-plugins libgz-tools2-dev libgz-transport12 libgz-transport12-core-dev
  libgz-transport12-dev libgz-transport12-log libgz-transport12-log-dev libgz-transport12-parameters libgz-transport12-parameters-dev libgz-utils2
  libgz-utils2-cli-dev libgz-utils2-dev libogre-next-2.3-dev libogre-next-2.3.0 libsdformat13 libsdformat13-dev ros-iron-actuator-msgs
  ros-iron-compressed-depth-image-transport ros-iron-compressed-image-transport ros-iron-image-transport-plugins ros-iron-ros-gz ros-iron-ros-gzgarden-bridge
  ros-iron-ros-gzgarden-image ros-iron-ros-gzgarden-interfaces ros-iron-ros-gzgarden-sim ros-iron-ros-gzgarden-sim-demos ros-iron-sdformat-urdf
  ros-iron-theora-image-transport ros-iron-xacro sdformat13-sdf
0 upgraded, 125 newly installed, 0 to remove and 1 not upgraded.
Need to get 0 B/33,1 MB of archives.
After this operation, 201 MB of additional disk space will be used.
Do you want to continue? [Y/n] 

```

mjcarroll, 2023-10-30:
Do you have any other PPAs or repositories installed?  To me, it looks like you are using one of the "unofficial" binaries that we have packaged here: https://build.osrfoundation.org/job/ros_gzgarden_bridge-install-pkg_iron-ci-jammy-amd64/4/

Those should not be aviailable on packages.ros.org by default, though.

wittenator, 2023-10-31:
I previously had Gazebo installed separately. Am I supposed to remove the Gazebo PPAs before trying to install Gazebo via ros-gz? As far as I see e.g. ros-iron-ros-gzgarden-image comes from the PPA: http://packages.osrfoundation.org/gazebo/ubuntu-stable jammy/main.

mjcarroll, 2023-10-31:
When you want to use the ros_gz packages from a ros distribution (such as humble/iron/rolling), that will source the gazebo packages from `packages.ros.org`.  This is to provide maximum compatibility for ROS users without having to fuss with PPAs.  As a result, the gazebo packages may lag behind, as they are pinned to LTS versions (in this case fortress rather than garden).

The newest gazebo packages are always available from `packages.osrfoundation.org`, which is the package mirror for gazebo.

The `ros-iron-ros-gz-garden` package that you used from `packages.osrfoundation.org` is an unofficial convenience package that we provide for users who want to use unsupported binary combinations.  This allows you to skip building the bridge from source.

azeey, 2023-10-31:
I think `ros-iron-ros-gz` should only install the version of Gazebo in Fortress. The unofficial versions have a suffix (`gzgarden` or `gzharmonic`). I just did a quick check with `apt-cache show ros-iron-ros-gz` and I get
```
Package: ros-iron-ros-gz
Version: 0.245.0-1002jammy
Architecture: amd64
Maintainer: Louise Poubel <louise@openrobotics.org>
Installed-Size: 43
Depends: ros-iron-ros-gzgarden-bridge, ros-iron-ros-gzgarden-image, ros-iron-ros-gzgarden-sim, ros-iron-ros-gzgarden-sim-demos, ros-iron-ros-workspace
Priority: optional
Section: misc
Filename: pool/main/r/ros-iron-ros-gz/ros-iron-ros-gz_0.245.0-1002jammy_amd64.deb
Size: 6790
SHA256: 8abbdbe70724de061007ae656817f53915e4846a76fff9b49949b1e6da428b7e
SHA1: bbeee5a86b9e0f4787d6ab2760c4ff16f1a6ccb0
MD5sum: 0aa953b8e75e91204ea4c8cde5609f77
Description: Meta-package containing interfaces for using ROS 2 with Gazebo simulation.
Description-md5: 7c163df521e3acf63ef2167539969298

Package: ros-iron-ros-gz
Version: 0.245.0-1jammy.20230908.202510
Architecture: amd64
Maintainer: Louise Poubel <louise@openrobotics.org>
Installed-Size: 43
Depends: ros-iron-ros-gz-bridge, ros-iron-ros-gz-image, ros-iron-ros-gz-sim, ros-iron-ros-gz-sim-demos, ros-iron-ros-workspace
Priority: optional
Section: misc
Filename: pool/main/r/ros-iron-ros-gz/ros-iron-ros-gz_0.245.0-1jammy.20230908.202510_amd64.deb
Size: 6892
SHA256: e6ce382ed8f33f5c5962156208ca90c0e38f0a5a2c78781ed9d36b8904078bce
SHA1: 57a3160cd836161859837be90a22d8aa1fa7be88
MD5sum: 97e5e3215bb308d975f67d0dd342495b
Description: Meta-package containing interfaces for using ROS 2 with Gazebo simulation.
Description-md5: 7c163df521e3acf63ef2167539969298
```

And `apt-cache policy ros-iron-ros-gz` shows:
```
ros-iron-ros-gz:
  Installed: (none)
  Candidate: 0.245.0-1002jammy
  Version table:
     0.245.0-1002jammy 500
        500 http://packages.osrfoundation.org/gazebo/ubuntu-prerelease jammy/main amd64 Packages
     0.245.0-1jammy.20230908.202510 500
        500 http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages

```

This shows that there's a `ros-iron-ros-gz` in the `ubuntu-prerelease` repo that brings us Garden. @wittenator do you happen to have prereleases enabled on your system?

j-rivero, 2023-10-31:
> Installing ros-iron-ros-gz installs Gazebo Garden instead of Fortress

This is a bug if this it is happening combining `{ros,osrfoundation}.packages.org`.

> I think `ros-iron-ros-gz` should only install the version of Gazebo in Fortress. The unofficial versions have a suffix (`gzgarden` or `gzharmonic`). I just did a quick check with `apt-cache show ros-iron-ros-gz` and I get

This is correct.

> Anyways, we should remove the version in our prereleases to avoid any confusion. cc @j-rivero

Seems a bug indeed while I was producing the prerelease packages. I would prefer not to remove packages but override them with newer versions so the upgrade path is easier. There are some changes in `ros_gz` upstream that we can potentially release.

wittenator, 2023-11-02:
The weird thing is that I didn't enable pre-releases on my system.

j-rivero, 2023-11-03:
> The weird thing is that I didn't enable pre-releases on my system. If I run `apt-cache depends ros-iron-ros-gz`, I also get this:
> `ros-iron-ros-gz Depends: ros-iron-ros-gzgarden-bridge Depends: ros-iron-ros-gzgarden-image Depends: ros-iron-ros-gzgarden-sim Depends: ros-iron-ros-gzgarden-sim-demos Depends: ros-iron-ros-workspace `,

This is wrong indeed. Could you please attach: `apt-cache policy ros-iron-ros-gz`

wittenator, 2023-11-04:
I missed the pre-release PPA in my sources list. After removing the offending PPA, I indeed get 
`ros-iron-ros-gz:
  Installed: 0.245.0-1jammy.20230908.202510
  Candidate: 0.245.0-1jammy.20230908.202510
  Version table:
 *** 0.245.0-1jammy.20230908.202510 500
        500 http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages
        100 /var/lib/dpkg/status
` as an output of `apt-cache policy ros-iron-ros-gz`

and the output of `apt-cache depends ros-iron-ros-gz` is
`ros-iron-ros-gz
  Depends: ros-iron-ros-gz-bridge
  Depends: ros-iron-ros-gz-image
  Depends: ros-iron-ros-gz-sim
  Depends: ros-iron-ros-gz-sim-demos
  Depends: ros-iron-ros-workspace`

I think that having the prerelease PPA active and trying to install `ros-iron-ros-gz` without any modifier should not install the harmonic or garden version despite this, right?