gazebosim/ros_gz #455: ros_gz_bridge can't compile on humble/garden: rosidl_pycommon on Ubuntu 22.04, Win10 WSL2

Twenkid, 2023-10-18:
Hi, I'm trying to use ros_gz bridge, but I face two problems. 

The bridge from the ros2 installation works, but only when a simulation is called with: "ign gazebo" as in the examples here, for sensors_demo.

If the same demo is invoked with gz sim, the topic appears in gz topic -l and ros2 topic list, but there's no image in rqt_view_image.

I tried to build it from source and tried to follow all instructions and tips I could find, cloning the humble branch etc., setting GZ_VERSION=garden etc., but so far it always ends up there:

**ModuleNotFoundError: No module named 'rosidl_pycommon'**

```
Traceback (most recent call last):
  File "/home/tosh/ros_gz_ws/src/ros_gz/ros_gz_bridge/bin/ros_gz_bridge_generate_tests", line 25, in <module>
    from ros_gz_bridge import generate_test_cpp
  File "/home/tosh/ros_gz_ws/src/ros_gz/ros_gz_bridge/ros_gz_bridge/__init__.py", line 21, in <module>
    from rosidl_pycommon import expand_template
ModuleNotFoundError: No module named 'rosidl_pycommon'
Traceback (most recent call last):
  File "/home/tosh/ros_gz_ws/src/ros_gz/ros_gz_bridge/bin/ros_gz_bridge_generate_tests", line 25, in <module>
    from ros_gz_bridge import generate_test_cpp
  File "/home/tosh/ros_gz_ws/src/ros_gz/ros_gz_bridge/ros_gz_bridge/__init__.py", line 21, in <module>
    from rosidl_pycommon import expand_template
ModuleNotFoundError: No module named 'rosidl_pycommon'
Traceback (most recent call last):
  File "/home/tosh/ros_gz_ws/src/ros_gz/ros_gz_bridge/bin/ros_gz_bridge_generate_tests", line 25, in <module>
    from ros_gz_bridge import generate_test_cpp
  File "/home/tosh/ros_gz_ws/src/ros_gz/ros_gz_bridge/ros_gz_bridge/__init__.py", line 21, in <module>
    from rosidl_pycommon import expand_template
ModuleNotFoundError: No module named 'rosidl_pycommon'
gmake[2]: *** [CMakeFiles/test_ros_publisher.dir/build.make:80: generated/test/gz_publisher.cpp] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:421: CMakeFiles/test_ros_publisher.dir/all] Error 2
gmake[2]: *** [CMakeFiles/test_gz_subscriber.dir/build.make:80: generated/test/gz_publisher.cpp] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:448: CMakeFiles/test_gz_subscriber.dir/all] Error 2
gmake[2]: *** [CMakeFiles/test_gz_publisher.dir/build.make:80: generated/test/gz_publisher.cpp] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:394: CMakeFiles/test_gz_publisher.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ros_gz_bridge [1.27s, exited with code 2]

Summary: 2 packages finished [5.00s]
  1 package failed: ros_gz_bridge
  1 package had stderr output: ros_gz_bridge
  3 packages not processed
  ```
  
  Also ROS documentation suggests that this module is not supported for humble:
   https://index.ros.org/p/rosidl_pycommon/#humble
   
   The rosidl ... pkgs are installed in ROS, I can't find "rosidl_pycommon" in particular, some py binding or something
   
   https://github.com/ros2/rosidl
   
Here:   
https://repo.ros2.org/status_page/humble_rhel.html

I see only "ros-rolling-rosidl-pycommon | 3.4.0-1"  

Is humble and gz sim actually supported anymore and what's a possible workaround or switching to Rolling is the solution?

(One I saw is capturing the gz messages I want from another program and posting them myself to ROS, or just the ROS node capturing them using gazebo code or something, but I don't know yet how much work would that involve.)

Thanks.
  

## Environment
* OS Version:  22.04 WSL
* Source, cloned from here, branch humble. The current now. Possibly: https://github.com/gazebosim/ros_gz/commit/9b0e3a9c31a9028541ee279677d572b635076f2d
  
## Description
* Expected behavior: A complete build.
* Actual behavior: Fails due to a missing dependency.

## Steps to reproduce

The instructions from this repo.

## Output

See above.

Twenkid, 2023-10-18:
I tried a partial workaround:

```ros2 run ros_gz_image image_bridge /rgbd_camera/image /rgbd_camera/image```

The topics appear in rqt_view

[image]

* However without images with gz sim: 

gz sim -v --render-engine ogre sensors_demo.sdf

...

* WIth ign gazebo the images show up:

ign gazebo -v --render-engine ogre sensors_demo.sdf

[image]

Twenkid, 2023-10-18:
Maybe I figured it out - I checked again, I discovered that git branch displayed ros2, IDK why. Re-cloned properly and this time it did complete (it takes a lot of RAM/page file though).

Then it didn't work until I read the bridge command from the example carefully - I noticed that the declared type was for Ignition, that's why it works only with ign...

It should be gz.msgs.Image, not ignition.msg.Image.

```
ros2 run ros_gz_bridge parameter_bridge /rgbd_camera/depth_image@sensor_msgs/msg/Image@gz.msgs.Image
```

Please update the examples. Thanks

mjcarroll, 2023-11-09:
>  (it takes a lot of RAM/page file though)

Yes, there are a lot of things happening at compile time with templates.  The binaries are suggested if you are on a limited platform.

> It should be gz.msgs.Image, not ignition.msg.Image.

Depending on the branch, it should be compatible with both.  Which branch did you end up using?

Twenkid, 2023-11-11:
Right, binary is easier, but there wasn't such a combination listed for my case: **humble + garden.** As of RAM I had plenty (32 GB), but I use to fill it with stuff and RAM disks. I mentioned that for anyone else who encounters WSL crashes while building.

azeey, 2024-01-04:
I don't think there's any remaining issue here, so I'll go ahead and close it

closed 2024-01-04 by azeey