gazebosim/ros_gz #456: Pose frames are lost in transport

simutisernestas, 2023-10-20:
## Environment
* OS Version: 
Humble distro, from APT

## Description
* Actual behavior: 
Gazebo pose messages when transported to ROS tf messages lose the frame info. For instance this is one of the gazebo messages:
```
pose {
  name: "x500_0"
  id: 25
  position {
    x: -24.332399998778531
    y: -35.4081
    z: 0.77342682381223771
  }
  orientation {
    x: -4.686981005291697e-17
    y: 2.72876828650036e-09
    z: -1.9894898985671598e-17
    w: 1
  }
}
```
And at the other end only the pose info is passed, but the frame names are lost.

* Expected behavior:
In the end would be nice to have static transform names passed, in this way for use in ROS applications. 

## Steps to reproduce

1. Fire a up a simulation with an entity
2. Launch transport
3. Check tf messages on ROS side
4. The frame name infromation is lost

azeey, 2023-11-10:
The steps you listed to reproduce the problem aren't very clear. Which launch files are you using? Is there a particular repository with an example I can look at?

simutisernestas, 2023-11-11:
You can simply launch any world that has pose information; for instance NAO joint control - included by default with gz installation when you launch `gz sim` and choose that world;

Create a `topic_map.yml` with contents:
```yaml
- ros_topic_name: "/gz/pose_array"
  gz_topic_name: "/world/nao_world/pose/info"
  ros_type_name: "geometry_msgs/msg/Pose"
  gz_type_name: "gz.msgs.Pose_V"
```
Launch bridge:
```ros2 run ros_gz_bridge parameter_bridge --ros-args -p config_file:=topic_map.yml```

The frame ids are not passed to ROS; If you look at the topic it's just a bunch of poses that you need to index manually to get the actual transforms, if simulation adds a new link all of these need to updated again. This is also related to `sdf` and `urdf` inter-op, most likely people have better solution for the use case. 

So for example at that time I was trying to get static tf's from gazebo to transform some sensor data to base_link and this is a hassle currently because the frame information is lost in the transport. Of course pose array has only single `frame_id` included while gazebo published all a list with `id` attached to each of them. Need some consideration how to implement this properly and a test could be creating a joint/static tf publisher on ROS side. 

Could help out with the feature, just need some input from your side how to go about it in a nice way. @azeey

azeey, 2024-01-04:
Take a look at [this](https://github.com/gazebosim/ros_gz/blob/ros2/ros_gz_sim_demos/launch/tf_bridge.launch.py) example. For `TF`, you'll need to change the `ros_type_name` to `tf2_msgs/msg/TFMessage`. You should then be able to publish it directly to `/tf` by changing `ros_topic_name` to `/tf`. That example depends on the `PosePublisher` system. [This](https://github.com/gazebosim/ros_gz/blob/ros2/ros_gz_sim_demos/launch/sdf_parser.launch.py) other example might also be useful.

Rendok, 2024-02-04:
I believe not only `frame_id`s are missing but also the time stamps. I observe the same behavior on ROS2 Humble and Gazebo Harmonic.

I tried `ros2 run ros_gz_bridge parameter_bridge /world/empty_world/pose/info@tf2_msgs/msg/TFMessage@gz.msgs.Pose_V`
but still, observe the same missing frame_ids and time stamps

```
- header:
    stamp:
      sec: 0
      nanosec: 0
    frame_id: ''
  child_frame_id: ''
  transform:
    translation:
      x: 0.0
      y: 0.0
      z: 10.0
    rotation:
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
```

ns9, 2024-02-14:
I am also experiencing this issue, is there any way to get the frame_id's to make their way to the ROS messages with the current implementation of the bridge or is a PR necessary to fix this?

azeey, 2024-02-14:
I think the problem is `/world/<world name>/pose/info` was not meant to be used with ROS. You'll want to set up the PosePublisher system.

This is actually a duplicate of https://github.com/gazebosim/ros_gz/issues/172, so I'll go ahead and close it.

closed 2024-02-14 by azeey

ns9, 2024-02-15:
@azeey Thanks for the quick response, I'm having issues getting the PosePublisher to produce world frame poses (I just see child links poses of the model) but I'll ask my question in the other open issue.