gazebosim/ros_gz #457: 2D Bounding Box messages from bounding box cameras are not bridged

wittenator, 2023-10-27:
## Desired behavior
There should be a bridge for the translation of 2D BBox messages coming from bounding box cameras.

## Alternatives considered
None

## Implementation suggestion
The corresponding ignition messages should be added. I did this in https://github.com/wittenator/ros_gz/tree/humble .

## Additional context
This is useful if one wants to replace e.g. an object detection node that runs on CUDA-hardware only with a Gazebo bounding box camera to make the simulation run on any hardware.

wittenator, 2023-11-09:
@ahcorde Is the port to ROS Iron something the Ros_gz team will do or should I do another PR?

ahcorde, 2023-11-09:
@wittenator, if you want to take care of the forward port I really appreciate it otherwise I will do it this or next week.

wittenator, 2023-11-09:
Sure, I don't mind doing that. I'll probably come around to it on Monday tho

ahcorde, 2023-11-09:
I can make a release today, and changes will be available in the next humble sync.

wittenator, 2023-11-09:
The humble release is not urgent for me anymore, I switched to Iron in the meantime^^ I was just curious, but thanks for the info!

mjcarroll, 2023-11-20:
@ahcorde to close after rolling release

ahcorde, 2023-11-21:
PR was forward ported to `rolling` too https://github.com/gazebosim/ros_gz/pull/466. 

Closed. Thank you @wittenator for your contributions.

PR #458 (2023-11-09): Added messages for 2D Bounding Boxes to ros_gz_bridge
# 🎉 New feature

Closes #457

## Summary
I added the bridging of Detection2D and Detection2DArray to AnnotatedAxisAligned2DBox and AnnotatedAxisAligned2DBox_V and vice versa. 

## Test it
WIP

## Checklist
- [X] Signed all commits for DCO
- [X] Added tests
- [ ] Added example and/or tutorial
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [X] `codecheck` passed (See [contributing](https://gazebosim.org/docs/all/contributing#contributing-cod [... cut]
closed 2024-01-07 by wittenator