gazebosim/ros_gz #459: Make `ros_gz_sim::create` use ROS parameters

mjcarroll, 2023-11-09:
Currently, the `ros_gz_sim::create` node uses gflags to configure where and how models are loaded.

For maximum compatibility with ROS, I think it makes sense to switch the gflags arguments to use rclcpp::Parameters.

ahcorde, 2023-11-14:
Hi @yaswanth1701, all the gflags arguments

Ayush1285, 2023-11-14:
There is an gflag argument 'param', which allowed us to pass any ros parameters which is not declared in the create.cpp file. So, now that we are converting gflag arguments to Parameters itself, we will lose this functionality, right?

Ayush1285, 2023-11-15:
@mjcarroll any comments?

mjcarroll, 2023-11-16:
@Ayush1285 In my opinion, since this is currently primarily used from ROS launch files, it would make sense to port all available parameters to be ROS parameters.  In the case that no ROS parameter arguments are available, we can revert to the `gflags` behavior.  I think preserving behavior is also important for the existing launch files that are out there.

An alternative to the gflags "fallback" parsing is to provide two separate executables, one exclusively gflags and one exclusively ROS paramaters, but I think that would result in a good bit of duplication.  I'll leave that up to whoever implements it to see which way works better.

I'm envisioning getting us positioned to take advantage of custom launch actions down the road, so the more "native" ROS support we have, the better.

Ayush1285, 2023-11-16:
I'm thinking about implementing this. I'll go with the first behaviour.

PR #475 (2023-12-28): Added support for using ROS 2 parameters to spawn entities in Gazebo using ros_gz_sim::create
# 🎉 New feature

Closes #459 

## Summary
Added ROS 2 parameters support in ros_gz_sim::create node, while preserving the existing behavior. ROS parameters will be checked first, if not found then will revert to Gflags arguments.

## Test it
Launch the demo launch file from the ros_gz_sim_demos package.
`ros2 launch ros_gz_sim_demos robot_description_publisher.launch.py`

## Checklist
- [x] Signed all commits for DCO
- [ ] Added tests
- [ ] Added example and/or tutorial
- [ ] Updated documentation (as [... cut]
closed 2023-12-28 by ahcorde