gazebosim/ros_gz #461: ros_gz for humble/harmonic

0mnimon, 2023-11-14:
As per gazebosim doc it is mentioned that you can download ros_gz for humble/harmonic using "ros-humble-ros-gzharmonic"
I tried both prerelease and released repo but both are providing error that "the command does not exist"
How do I get ros_gz for humble/harmonic?

[image]

azeey, 2023-11-14:
@0mnimon, thanks for the issue. It's an error in the documentation. Currently the Humble/Harmonic combination is not supported.

azeey, 2023-11-14:
https://github.com/gazebosim/ros_gz/pull/462

closed 2023-11-20 by mjcarroll

quantumxt, 2024-09-18:
Hi there, just wondering if there's a binary for `arm64` architecture? 

I compared the following Packages list:
- `amd64` (https://packages.osrfoundation.org/gazebo/ubuntu-stable/dists/jammy/main/binary-amd64/) 
- `arm64` (https://packages.osrfoundation.org/gazebo/ubuntu-stable/dists/jammy/main/binary-arm64/)

`ros-humble-ros-gzharmonic` was only listed inside `amd64`, but not `arm64`.