gazebosim/ros_gz #469: gz_sim create subscribes to wrong topic

muttistefano, 2023-12-12:
hi all, 
i am on Ubuntu 22.04, Ros2 rolling, with ros_gz built from source.

I have noticed that the create node of ros_gz_sim has a weird behaviour.
when i run : 
`ros2 run ros_gz_sim create -topic sweepee_1/robot_description -name sweepee_2`

the node subscribe to the topic "sweepee": 
`[INFO] [1702368228.031667490] [ros_gz_sim]: Waiting messages on topic [sweepee].`

Whenever i change topic, the node subscribe only to the first 7 letters of the provided string.

```
ros2 run ros_gz_sim create -topic abcdefghil  
[INFO] [1702368480.765282089] [ros_gz_sim]: Requesting list of world names.  
[INFO] [1702368480.966820664] [ros_gz_sim]: Waiting messages on topic [abcdefg].  
```
Did anyone experience this ?
Thanks

ahcorde, 2023-12-12:
I can't reproduce the issue, I tried from source and debs. 

Can you provide more information ?

muttistefano, 2023-12-12:
Debs also work fine in my case, but being on rolling i cannot use them.
Checking the source i can say that the FLAGS_topic variable seems to be the problem , but it is generated and read by the gflags library.
I indeed have the correct gflags library installed so i don't really know what's  wrong. 
I can try to reporoduce the problem in a new machine and check if it applies.
Thanks

muttistefano, 2023-12-22:
commit 8fba3b4 solved the problem.

closed 2023-12-22 by muttistefano