gazebosim/ros_gz #474: Unable to Source install ros-gz Branch ros2 on Ubuntu 22.04 ros2 humble Gazebo Harmonic (ver8)

ARLunan, 2023-12-24:
## Environment
* OS Version: Ubuntu 22.04 virtual machine on Host MAC M1 arm64
*Binary Branch ros2 branch 
* All Harmonic required gz-sim libraries installed from Binary

## Description 
Actual behavior: ros_gz_sim failed , Error missing gz-transport12 which is installed but for for gz-garden . How to force gz-harmonic ?

azeey, 2024-01-02:
For building `humble` from source, you'll need to use the `humble` branch of `ros_gz`. Can you give that a try?

ARLunan, 2024-01-02:
OK tnx: Here goes on my UB22.04/arm64 VM on a MAC M1 with ROS 2 humble & gz_sim harmonic installed. Here's a copy of the relevant transcription of the colcon build. While compile did not complete, it seems the previously reported rosicl-pycommon error dod not occur.
ubuntutbc@macvm-ub22h-bv:~/rosgz_ws$ gz sim --version
Gazebo Sim, version 8.0.0
Copyright (C) 2018 Open Source Robotics Foundation.
Released under the Apache 2.0 License.

ubuntutbc@macvm-ub22h-bv:~/rosgz_ws
$ ubuntutbc@macvm-ub22h-bv:~/rosgz_ws/src/ros_gz/ros_gz$ git branch
* humble
  ros2
ubuntutbc@macvm-ub22h-bv:~/rosgz_ws/src/ros_gz/ros_gz$
ubuntutbc@macvm-ub22h-bv:~/rosgz_ws$ rosdep install -r --from-paths src -i -y --rosdistro humble
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ros_gz_image: Cannot locate rosdep definition for [gz-transport13]
ros_gz_bridge: Cannot locate rosdep definition for [gz-transport13]
ros_gz_sim: Cannot locate rosdep definition for [gz-transport13]
Continuing to install resolvable dependencies...
......
#All required rosdeps installed successfully
ubuntutbc@macvm-ub22h-bv:~/rosgz_ws$ gz sim --version
Gazebo Sim, version 8.0.0
Copyright (C) 2018 Open Source Robotics Foundation.
Released under the Apache 2.0 License.

ubuntutbc@macvm-ub22h-bv:~/rosgz_ws$ ros2 pkg list|grep sim
nav2_simple_commander
rviz_assimp_vendor
ubuntutbc@macvm-ub22h-bv:~/rosgz_ws$ ros2 pkg list|grep gz
ubuntutbc@macvm-ub22h-bv:~/rosgz_ws$ ros2 pkg list|grep gazebo
ubuntutbc@macvm-ub22h-bv:~/rosgz_ws$ colcon build
Starting >>> ros_gz_interfaces
Starting >>> ros_gz_sim
Finished <<< ros_gz_interfaces [2.21s]                                      
Starting >>> ros_gz_bridge
Finished <<< ros_gz_sim [11.4s]                                        
Starting >>> ros_ign_gazebo
Finished <<< ros_ign_gazebo [5.29s]                                      
Starting >>> ros_ign_interfaces
[Processing: ros_gz_bridge, ros_ign_interfaces]                          
Finished <<< ros_ign_interfaces [38.3s]                                  
[Processing: ros_gz_bridge]                                        
[Processing: ros_gz_bridge]                                        
[Processing: ros_gz_bridge]                                        
[Processing: ros_gz_bridge]                                        
[Processing: ros_gz_bridge]                                        
[Processing: ros_gz_bridge]                                        
[Processing: ros_gz_bridge]                                        
[Processing: ros_gz_bridge]                                        
[Processing: ros_gz_bridge]                                        
[Processing: ros_gz_bridge]                                        
[Processing: ros_gz_bridge]] [ros_gz_bridge:build 43% - 6min 25.1s]
[Processing: ros_gz_bridge]                                        
[Processing: ros_gz_bridge]                                        
[Processing: ros_gz_bridge]                                        
[Processing: ros_gz_bridge]                                        
[Processing: ros_gz_bridge]                                        
[Processing: ros_gz_bridge]                                        
[Processing: ros_gz_bridge]                                        
[Processing: ros_gz_bridge]                                          
[Processing: ros_gz_bridge]                                          
[Processing: ros_gz_bridge]                                          
[Processing: ros_gz_bridge]                                          
[Processing: ros_gz_bridge]                                          
[Processing: ros_gz_bridge]                                          
[Processing: ros_gz_bridge]e] [ros_gz_bridge:build 43% - 13min 29.4s]
[Processing: ros_gz_bridge]                                          
[Processing: ros_gz_bridge]e] [ros_gz_bridge:build 43% - 14min 29.0s]
[Processing: ros_gz_bridge]                                          
[Processing: ros_gz_bridge]                                                                     
[... 16 lines cut ...]
[Processing: ros_gz_bridge]                                          
[Processing: ros_gz_bridge]                                          
[Processing: ros_gz_bridge]                                          
[Processing: ros_gz_bridge]                                          
[Processing: ros_gz_bridge]                                          
[Processing: ros_gz_bridge]                                          
[Processing: ros_gz_bridge]                                          
[Processing: ros_gz_bridge]                                          
[Processing: ros_gz_bridge]                                          
[Processing: ros_gz_bridge]                                          
[Processing: ros_gz_bridge]                                          
--- stderr: ros_gz_bridge                                            
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/ros_gz_bridge_lib.dir/build.make:497: CMakeFiles/ros_gz_bridge_lib.dir/generated/factories/geometry_msgs.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:183: CMakeFiles/ros_gz_bridge_lib.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ros_gz_bridge [29min 36s, exited with code 2]

Summary: 4 packages finished [29min 39s]
  1 package failed: ros_gz_bridge
  1 package had stderr output: ros_gz_bridge
  7 packages not processed
ubuntutbc@macvm-ub22h-bv:~/rosgz_ws$

azeey, 2024-01-02:
Okay, that looks like you're running out of memory. Others have run into that before as well (https://github.com/gazebosim/ros_gz/issues/401#issuecomment-1568882258). Try adding `--parallel-workers=1 --executor sequential` to your `colcon` command.

ARLunan, 2024-01-02:
OK here is that trial:
ubuntutbc@macvm-ub22h-bv:~/rosgz_ws$ colcon build --parallel-workers=1 --executor sequential
Starting >>> ros_gz_interfaces
Finished <<< ros_gz_interfaces [0.88s]                      
Starting >>> ros_gz_sim
Finished <<< ros_gz_sim [0.28s]                     
Starting >>> ros_gz_bridge
--- stderr: ros_gz_bridge                                          
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/ros_gz_bridge_lib.dir/build.make:497: CMakeFiles/ros_gz_bridge_lib.dir/generated/factories/geometry_msgs.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:183: CMakeFiles/ros_gz_bridge_lib.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ros_gz_bridge [1min 51s, exited with code 2]

Summary: 2 packages finished [1min 53s]
  1 package failed: ros_gz_bridge
  1 package had stderr output: ros_gz_bridge
  9 packages not processed
ubuntutbc@macvm-ub22h-bv:~/rosgz_ws$

azeey, 2024-01-03:
I still think you're running out of memory. Can you monitor the memory usage during the build with a tool like `htop`?

ARLunan, 2024-01-03:
Here is htop display during the compile - Yes it appears memory capacity is a problem. There are 4 cores and all are 85 -100%
[image]

Since my Ubuntu system is virtual I will attempt to allocate more to VM/less to host.

ARLunan, 2024-01-03:
I doubled the Memory from 2 MB to 4 MB- still maxes out. The MAC has 4 Cores & 8 GB memory. Il  increase the Swap & allocate 1 more core & 6 GB.

azeey, 2024-01-03:
From [this](https://robotics.stackexchange.com/a/97898/31574) answer, it seems like colcon will use all available cores to build a single package, so you'll also need to do `export MAKEFLAGS="-j 1"` before running `colcon build`.

ARLunan, 2024-01-03:
azeey Success compiling ros_gz from Source : Summary: 12 packages finished [37 secs] with the allocating 4 Cores CPU /8 GB Memory and export MAKEFLAGS="-j 1". **Thank you**. I ran ROS Integration > ros_gz_bridge demo > Bidirectional Communication Demo . I ran the Visualize in RViz which opened but generated an error that I will troubleshoot as the file vehicle.sdf "missing" is located in the necessary Share directory . **ubuntutbc@macvm-ub22h-bv:~/rosgz_ws/install/ros_gz_sim_demos/share/ros_gz_sim_demos/worlds$ ls
vehicle.sdf**

ruby $(which gz) sim-1] [Err] [Server.cc:198] Error Code 14: [/sdf/world[@name="demo"]/model[@name="vehicle"]/include[0]/uri:/home/ubuntutbc/rosgz_ws/install/ros_gz_sim_demos/share/ros_gz_sim_demos/worlds/vehicle.sdf:L64]: Msg: Unable to find uri[package://ros_gz_sim_demos/models/vehicle]

So the ros_gz Harmonic (ros2 branch) Install on a ROS 2 Humble from Source on a suitably powerful arm64 machine works . Thank you for your assistance.

azeey, 2024-01-03:
Glad to hear it worked. For the issue with finding the file, make sure you've sourced your colcon workspace.

https://github.com/gazebosim/ros_gz/blob/ros2/ros_gz_sim_demos%2Fhooks%2Fros_gz_sim_demos.dsv.in contains `IGN_GAZEBO_RESOURCE_PATH`. It might need to be updated to `GZ_SIM_RESOURCE_PATH`.

ARLunan, 2024-01-04:
Implementing your indicated change and following https://gazebosim.org/docs/harmonic/ros2_integration ,**Visualize in RViz,** the RViz & Gz Windows open displaying the vehicle. But they don't respond to published /cmd_vel messages, either from the sample "move in circle" msg or running teleop_twist_keyboard in a separate terminal. With ros2 topic echo /cmd_vel in another terminal, I verified that /cmd_vel are being published. All the new terminal instances are sourced by a /opt/ros/humble and rosgz_ws statements in the .bashrc and the topics and nodes can be listed.