gazebosim/ros_gz #487: I want to run gazebo controllers with harmonic and humble

Apurv354, 2024-01-18:
## Environment
* OS Version: Ubuntu 22.04 
* Source or binary build?
  
  
Binary Installation for ROS2 humble + Source installation of ros_gz (humble branch) + Source installation of [gazebo_ros_control](https://github.com/ros-controls/gazebo_ros2_control) master branch

## Description
* Expected behavior: Trying to run an example script - `ros2 launch gz_ros2_control_demos cart_example_position.launch.py`
This should happen - https://control.ros.org/master/doc/gz_ros2_control/doc/index.html
* Actual behavior: 
```
apurv@apurv-Alienware-m15-R2:~/artefacts_ws/src$ ros2 launch gz_ros2_control_demos cart_example_position.launch.py 
[INFO] [launch]: All log files can be found below /home/apurv/.ros/log/2024-01-18-15-20-10-577185-apurv-Alienware-m15-R2-52537
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ruby $(which gz) sim-1]: process started with pid [52538]
[INFO] [robot_state_publisher-2]: process started with pid [52540]
[INFO] [create-3]: process started with pid [52543]
[create-3] [INFO] [1705558810.756671381] [ros_gz_sim]: Requesting list of world names.
[robot_state_publisher-2] [INFO] [1705558810.784014225] [robot_state_publisher]: got segment cart
[robot_state_publisher-2] [INFO] [1705558810.784120816] [robot_state_publisher]: got segment slideBar
[robot_state_publisher-2] [INFO] [1705558810.784130449] [robot_state_publisher]: got segment world
[ruby $(which gz) sim-1] 
[ruby $(which gz) sim-1] Using deprecated environment variable [IGN_GAZEBO_RESOURCE_PATH].
[ruby $(which gz) sim-1] Please use [GZ_SIM_RESOURCE_PATH] instead.
[ruby $(which gz) sim-1] [Dbg] [gz.cc:166] Subscribing to [/gazebo/starting_world].
[ruby $(which gz) sim-1] [Dbg] [gz.cc:168] Waiting for a world to be set from the GUI...
[ruby $(which gz) sim-1] [Msg] Received world [empty.sdf] from the GUI.
[ruby $(which gz) sim-1] [Dbg] [gz.cc:172] Unsubscribing from [/gazebo/starting_world].
[ruby $(which gz) sim-1] [Msg] Gazebo Sim Server v8.0.0
[ruby $(which gz) sim-1] [Msg] Loading SDF world file[/usr/share/gz/gz-sim8/worlds/empty.sdf].
[ruby $(which gz) sim-1] [Msg] Serving entity system service on [/entity/system/add]
[ruby $(which gz) sim-1] [Dbg] [Physics.cc:860] Loaded [gz::physics::dartsim::Plugin] from library [/usr/lib/x86_64-linux-gnu/gz-physics-7/engine-plugins/libgz-physics-dartsim-plugin.so]
[ruby $(which gz) sim-1] [Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::Physics] for entity [1]
[ruby $(which gz) sim-1] [Msg] Create service on [/world/empty/create]
[ruby $(which gz) sim-1] [Msg] Remove service on [/world/empty/remove]
[ruby $(which gz) sim-1] [Msg] Pose service on [/world/empty/set_pose]
[ruby $(which gz) sim-1] [Msg] Pose service on [/world/empty/set_pose_vector]
[ruby $(which gz) sim-1] [Msg] Light configuration service on [/world/empty/light_config]
[ruby $(which gz) sim-1] [Msg] Physics service on [/world/empty/set_physics]
[ruby $(which gz) sim-1] [Msg] SphericalCoordinates service on [/world/empty/set_spherical_coordinates]
[ruby $(which gz) sim-1] [Msg] Enable collision service on [/world/empty/enable_collision]
[ruby $(which gz) sim-1] [Msg] Disable collision service on [/world/empty/disable_collision]
[ruby $(which gz) sim-1] [Msg] Material service on [/world/empty/visual_config]
[ruby $(which gz) sim-1] [Msg] Material service on [/world/empty/wheel_slip]
[ruby $(which gz) sim-1] [Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::UserCommands] for entity [1]
[ruby $(which gz) sim-1] [Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::SceneBroadcaster] for entity [1]
[ruby $(which gz) sim-1] [Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::Contact] for entity [1]
[ruby $(which gz) sim-1] [Msg] Loaded level [3]
[ruby $(which gz) sim-1] [Msg] Serving world controls on [/world/empty/control], [/world/empty/control/state] and [/world/empty/playback/control]
[ruby $(which gz) sim-1] [Msg] Serving GUI information on [/world/empty/gui/info]
[ruby $(which gz) sim-1] [Msg] World [empty] initialized with [1ms] physics profile.
[ruby $(which gz) sim-1] [Msg] Serving world SDF generation service on [/world/empty/generate_world_sdf]
[ruby $(which gz) sim-1] [Msg] Serving world names on [/gazebo/worlds]
[ruby $(which gz) sim-1] [Msg] Resource path add service on [/gazebo/resource_paths/add].
[ruby $(which gz) sim-1] [Msg] Resource path get service on [/gazebo/resource_paths/get].
[ruby $(which gz) sim-1] [Msg] Resource path resolve service on [/gazebo/resource_paths/resolve].
[create-3] [ERROR] [1705558811.319355172] [ros_gz_sim]: Must specify either -file, -param, -stdin or -topic
[ERROR] [create-3]: process has died [pid 52543, exit code 255, cmd '/home/apurv/artefacts_ws/src/install/ros_gz_sim/lib/ros_gz_sim/create --ros-args --params-file /tmp/launch_params_ddwn3yon'].
[INFO] [ros2-4]: process started with pid [52610]
[ruby $(which gz) sim-1] [GUI] [Wrn] [Application.cc:904] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:52:3: QML RenderWindowOverlay: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[ruby $(which gz) sim-1] [GUI] [Wrn] [Application.cc:904] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:67:3: QML EntityContextMenu: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[ruby $(which gz) sim-1] [GUI] [Wrn] [Application.cc:904] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:52:3: QML RenderWindowOverlay: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[ruby $(which gz) sim-1] [GUI] [Wrn] [Application.cc:904] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:67:3: QML EntityContextMenu: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[ruby $(which gz) sim-1] [GUI] [Wrn] [Application.cc:904] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:52:3: QML RenderWindowOverlay: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[ruby $(which gz) sim-1] [GUI] [Wrn] [Application.cc:904] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:67:3: QML EntityContextMenu: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[ruby $(which gz) sim-1] [GUI] [Wrn] [Application.cc:904] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:52:3: QML RenderWindowOverlay: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[ruby $(which gz) sim-1] [GUI] [Wrn] [Application.cc:904] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:67:3: QML EntityContextMenu: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[ruby $(which gz) sim-1] [GUI] [Wrn] [Application.cc:904] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:52:3: QML RenderWindowOverlay: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[ruby $(which gz) sim-1] [GUI] [Wrn] [Application.cc:904] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:67:3: QML EntityContextMenu: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[... 149 lines cut ...]
[ruby $(which gz) sim-1] [Msg] Serving full state on [/world/empty/state]
[ruby $(which gz) sim-1] [Msg] Serving full state (async) on [/world/empty/state_async]
[ruby $(which gz) sim-1] [Msg] Publishing scene information on [/world/empty/scene/info]
[ruby $(which gz) sim-1] [Msg] Publishing entity deletions on [/world/empty/scene/deletion]
[ruby $(which gz) sim-1] [Msg] Publishing state changes on [/world/empty/state]
[ruby $(which gz) sim-1] [Msg] Publishing pose messages on [/world/empty/pose/info]
[ruby $(which gz) sim-1] [Msg] Publishing dynamic pose messages on [/world/empty/dynamic_pose/info]
[ruby $(which gz) sim-1] [Dbg] [EntityComponentManager.cc:1655] Updated state thread iterators: 6 threads processing around 2 entities each.
[ruby $(which gz) sim-1] [Dbg] [SignalHandler.cc:142] Received signal[2].
[ruby $(which gz) sim-1] [Dbg] [ServerPrivate.cc:115] Server received signal[2]
[... cut ...]
[ruby $(which gz) sim-1] [Dbg] [SimulationRunner.cc:569] Exiting postupdate worker thread (1)
[ERROR] [ruby $(which gz) sim-1]: process has died [pid 52538, exit code -2, cmd 'ruby $(which gz) sim  -r -v 4 empty.sdf --force-version 8'].
[INFO] [launch]: process[ruby $(which gz) sim-1] was required: shutting down launched system

```

I don't see the model spawning.

## Steps to reproduce

1. Get the above environment and run - ros2 launch gz_ros2_control_demos cart_example_position.launch.py

## Output

azeey, 2024-01-26:
I assume you meant https://github.com/ros-controls/gz_ros2_control instead of https://github.com/ros-controls/gazebo_ros2_control. I believe the `master` branch is making use of changes in`ros_gz` that are only available in `rolling` https://github.com/ros-controls/gz_ros2_control/pull/211, so you'll also need to use the `humble` branch for `gz_ros2_control` as well.

Apurv354, 2024-01-27:
I have tried the humble branch, and it gives me the same error that I posted above.

Apurv354, 2024-01-28:
Basically, this is the problem 

```
[ruby $(which gz) sim-1] [INFO] [1706429551.323616584] [controller_manager]: Loading controller 'joint_state_broadcaster'
[ruby $(which gz) sim-1] [ERROR] [1706429551.323757245] [controller_manager]: Loader for controller 'joint_state_broadcaster' (type 'joint_state_broadcaster/JointStateBroadcaster') not found.
[ruby $(which gz) sim-1] [INFO] [1706429551.323796786] [controller_manager]: Available classes:
[ruby $(which gz) sim-1] [INFO] [1706429551.323854688] [controller_manager]:   controller_manager/test_controller
[ruby $(which gz) sim-1] [INFO] [1706429551.323873620] [controller_manager]:   controller_manager/test_controller_failed_init
[ruby $(which gz) sim-1] [INFO] [1706429551.323884612] [controller_manager]:   controller_manager/test_controller_with_interfaces
[ruby $(which gz) sim-1] [INFO] [1706429551.323898151] [controller_manager]:   controller_manager/test_chainable_controller

```

ahcorde, 2024-01-29:
@Apurv354 you need to install the dependency 

```bash
sudo apt-get install ros-humble-joint-state-broadcaster
```

or run `rosdep`

closed 2024-01-29 by Apurv354

sudhirpratapyadav, 2024-12-30:
How did you solved this issue, I tried `humble` branch but it only has `ign_ros2_control` package and not `gz_ros2_control`. I build it but when it only works with `fortress` and not `harmonic`