gazebosim/ros_gz #498: Unable to complete turtlebot3_simulations - new-branch "Migrating ROS 2 Packages that use Gazebo Classic"

ARLunan, 2024-02-18:
## Environment
* OS Version: <Ubuntu 22.04 on MAC M1 arm64arm64 -->
* Source
turtlebot3_simulations 2.2.6 new_gazebo latest commit "last week i/.e Feb 10  

## Description
After following procedure in https://gazebosim.org/docs/harmonic/migrating_gazebo_classic_ros2_packages, successful rosdep, clone https://github.com/azeey/turtlebot3_simulations.git
*  Expected behavior: successful colcon build of turlebot3_simulations
* Actual behavior: 
* buntutbc@macvm-ub22h-bv:~/turtlebot3_ws$ colcon build --symlink-install
Starting >>> turtlebot3_fake_node
Starting >>> turtlebot3_gazebo                     
--- stderr: turtlebot3_gazebo                                              
CMake Error at CMakeLists.txt:23 (find_package):
  By not providing "Findgazebo.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "gazebo", but
  CMake did not find one.

  Could not find a package configuration file provided by "gazebo" with any
  of the following names:

    gazeboConfig.cmake
    gazebo-config.cmake

  Add the installation prefix of "gazebo" to CMAKE_PREFIX_PATH or set
  "gazebo_DIR" to a directory containing one of the above files.  If "gazebo"
  provides a separate development package or SDK, be sure it has been
  installed.

---
Failed   <<< turtlebot3_gazebo [0.38s, exited with code 1]
Aborted  <<< turtlebot3_fake_node [0.50s]

Summary: 0 packages finished [1.08s]
  1 package failed: turtlebot3_gazebo
  1 package aborted: turtlebot3_fake_node
  1 package had stderr output: turtlebot3_gazebo
  1 package not processed
ubuntutbc@macvm-ub22h-bv:~/turtlebot3_ws$

azeey, 2024-03-12:
I think the issue is that `rosdep` doesn't know how to install Gazebo classic. Did you get any errors when running `rosdep install`? You might have to install Gazebo classic manually.

ARLunan, 2024-03-12:
Yes, I concur, but we do not want nor need Classic on our machines. I thought the whole point of the migration activity was to deploy the actively supported Gazebo Harmonic and later versions on our systems, including to use existing tutorials such as Turtlebot3. I am also in the group of users with arm64 machines, I.e. Linux Virtual Desktop on a MAC M1 for which it is impossible to install Classic due missing depandancies. I am very pleased that I can install Harmonic gz-sim (binary install) and ros-gz (source install), and able to migrate my existing “Classic” gazebo robot simulations packages to use Harmonic. For example,  with assistance with other developers,  I have successfully done this with the popular linorobot2 robot packages. So is there something I could try to get this new TB3 package to compile & work with gz-sim / ros-gz Harminic?

azeey, 2024-03-12:
Ah, I see. I thought you were running the first part of the tutorial where it builds the Gazebo classic example just to check everything is working as expected. If you only need the new Gazebo version, you'll have to checkout the `new_gazebo` branch from https://github.com/azeey/turtlebot3_simulations.git. That branch wasn't the default branch, so when you did `git clone`, you got the original turtlebot3_simulation package with Gazebo classic. I went ahead and made `new_gazebo` the default branch now so just doing `git clone` should work from now on, but if you've already cloned it, you'll need to do checkout the `new_gazebo` branch manually.

ARLunan, 2024-03-12:
Thanks, but I confirmed that I believe my a colcon build was against the "new_gazebo" branch, see the log following. Did forget something before doing the build?

ubuntutbc@macvm-ub22h-bv:~/turtlebot3_ws/src$ ls
turtlebot3_simulations
ubuntutbc@macvm-ub22h-bv:~/turtlebot3_ws/src$ cd turtlebot3_simulations
ubuntutbc@macvm-ub22h-bv:~/turtlebot3_ws/src/turtlebot3_simulations$ git status
On branch new_gazebo
Your branch is up to date with 'origin/new_gazebo'.

nothing to commit, working tree clean
ubuntutbc@macvm-ub22h-bv:~/turtlebot3_ws/src/turtlebot3_simulations$ git branch
  master
* new_gazebo
ubuntutbc@macvm-ub22h-bv:~/turtlebot3_ws/src/turtlebot3_simulations$ git checkout new_gazebo
Already on 'new_gazebo'
Your branch is up to date with 'origin/new_gazebo'.
ubuntutbc@macvm-ub22h-bv:~/turtlebot3_ws/src/turtlebot3_simulations$ cd ..
ubuntutbc@macvm-ub22h-bv:~/turtlebot3_ws/src$ cd ..
ubuntutbc@macvm-ub22h-bv:~/turtlebot3_ws$ colcon build 
Starting >>> turtlebot3_fake_node
Starting >>> turtlebot3_gazebo
--- stderr: turtlebot3_gazebo                                            
CMake Error at CMakeLists.txt:23 (find_package):
  By not providing "Findgazebo.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "gazebo", but
  CMake did not find one.

  Could not find a package configuration file provided by "gazebo" with any
  of the following names:

    gazeboConfig.cmake
    gazebo-config.cmake

  Add the installation prefix of "gazebo" to CMAKE_PREFIX_PATH or set
  "gazebo_DIR" to a directory containing one of the above files.  If "gazebo"
  provides a separate development package or SDK, be sure it has been
  installed.

---
Failed   <<< turtlebot3_gazebo [0.31s, exited with code 1]
Aborted  <<< turtlebot3_fake_node [1.12s]                           

Summary: 0 packages finished [1.59s]
  1 package failed: turtlebot3_gazebo
  1 package aborted: turtlebot3_fake_node
  1 package had stderr output: turtlebot3_gazebo
  1 package not processed
ubuntutbc@macvm-ub22h-bv:~/turtlebot3_ws$

kamccall, 2025-01-05:
@ARLunan were you ever able to figure this out?!?  i'm having the EXACT same problem with ROS2 humble and (new) gazebo and i've been trying different things all day long to try to get it to work... but alas, still no luck. 
any advice or suggestions??

azeey, 2025-01-07:
This issue has been fixed. There were some remaining gazebo-classic bits that were commented out to fix the issue @ARLunan reported.

@kamccall if you're on using ROS 2 Humble, you need to make sure to check out the `new_gazebo` branch.

The `rolling` branch on https://github.com/azeey/turtlebot3_simulations/ has been updated to work with ROS 2 Jazzy and newer. It will not work on Humble

closed 2025-01-07 by azeey