gazebosim/ros_gz #508: How can I install ros_gz between ROS2-foxy and IgnationGazebo-Garden

jzl20, 2024-03-08:
## Desired behavior

I am installing the PX4 environment, and my computer system is Ubuntu 20.04,
I have installed the ROS2-foxy (because my system is Ubuntu 20.04, although ROS2-Humble is more perfect) and IgnitionGazebo-Garden by following PX4 official tutorial.
But when I plan to install the ros_gz, I find there is not suitable combination.

The combinations only include:
ROS2-Humble <->  IgnitionGazebo-Garden
ROS2-Foxy <->  IgnitionGazebo-Edifice

How can I build up the ros_gz bridge between 
 ROS2-Foxy  and IgnitionGazebo-Garden

## Alternatives considered

## Implementation suggestion

## Additional context

azeey, 2024-03-11:
As listed in https://gazebosim.org/docs/garden/ros_installation, this combination of ROS 2 Foxy and Gazebo Garden is not supported. Is it listed on the PX4 documentation as a possible combination?

azeey, 2024-12-06:
Closing since this is not a bug in ros_gz

closed 2024-12-06 by azeey