gazebosim/ros_gz #513: ros_gz_bridge build fails "gmake Error 2"

biagio-trimarchi, 2024-03-20:
## Environment
* OS Version:  Ubuntu 22.04
* Source or binary build?
From source, branch "origin/humble".

## Description
* Expected behavior: Compiling without issues
* Actual behavior: Build failed, I think due to "running out of memory"

## Steps to reproduce
1.  Download the source code from github and fork the "humble" branch
2. Build the package with "colcon build"

## Output
```
colcon build
Starting >>> ros_gz_interfaces
Starting >>> ros_gz_sim
Finished <<< ros_gz_sim [14.2s]
Starting >>> ros_ign_gazebo
Finished <<< ros_ign_gazebo [1.44s]
Finished <<< ros_gz_interfaces [16.9s]
Starting >>> ros_gz_bridge
Starting >>> ros_ign_interfaces
[Processing: ros_gz_bridge, ros_ign_interfaces]
Finished <<< ros_ign_interfaces [35.6s]
[Processing: ros_gz_bridge]
[Processing: ros_gz_bridge]
[Processing: ros_gz_bridge]
[Processing: ros_gz_bridge]
[Processing: ros_gz_bridge]
[Processing: ros_gz_bridge]
[Processing: ros_gz_bridge]] [ros_gz_bridge:build 58% - 4min 14.3s]
[Processing: ros_gz_bridge]
[Processing: ros_gz_bridge]
[Processing: ros_gz_bridge]
[Processing: ros_gz_bridge]
[Processing: ros_gz_bridge]] [ros_gz_bridge:build 58% - 6min 56.3s]
[Processing: ros_gz_bridge]                                                           
[Processing: ros_gz_bridge]
[Processing: ros_gz_bridge]
[Processing: ros_gz_bridge]
[Processing: ros_gz_bridge]
--- stderr: ros_gz_bridge
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/ros_gz_bridge_lib.dir/build.make:539: CMakeFiles/ros_gz_bridge_lib.dir/generated/factories/ros_gz_interfaces.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:183: CMakeFiles/ros_gz_bridge_lib.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ros_gz_bridge [10min 29s, exited with code 2]

Summary: 4 packages finished [10min 46s]
  1 package failed: ros_gz_bridge
  1 package had stderr output: ros_gz_bridge
  7 packages not processed
```

ahcorde, 2024-03-20:
can you try to run the colcon command with `--parallel-workers 1`

I think you are running out of memory

biagio-trimarchi, 2024-03-20:
It worked with "--parallel-workers 4", after closing tmux and any other program.

Still, I think the package should be made "lighter" to build.

closed 2024-03-20 by ahcorde