gazebosim/ros_gz #529: `expand_gz_topic_names` parameter does not work

elkuno213, 2024-04-07:
## Environment
* OS Version: Ubuntu 22.04
* Source or binary build? binary build by `apt`, version `iron`

## Description
* Expected behavior: when `expand_gz_topic_names` is true, the namespace on Gazebo should be changed.
* Actual behavior: the namespace does not change and this parameter can be found via `ros2 param list`.

## Steps to reproduce

1. Follow the [Example 6](https://github.com/gazebosim/ros_gz/blob/ros2/ros_gz_bridge/README.md#example-6-using-ros-namespace-with-the-bridge).
2. Run this command:

```bash
ros2 run ros_gz_bridge parameter_bridge chatter@std_msgs/msg/String@ignition.msgs.StringMsg \
  --ros-args -p expand_gz_topic_names:=true -r __ns:=/demo
```

3. In other terminal, run this command:
```bash
ign topic --list 
# Output: /chatter
```
The topic is still `/chatter`, not `/demo/chatter` as expected. 

4. Run other command to check whether the parameter is existing:
 ```bash
ros2 param list | grep expand_gz_topic_names
# Output: nothing found
```

5. This [code snippet](https://github.com/gazebosim/ros_gz/blob/deccb5b9673896a4d9f6672ea9362ae1ec2f4dfa/ros_gz_bridge/src/ros_gz_bridge.cpp#L53) shows that `expand_gz_topic_names` only works when `config_file` is given.

## Output
[image]
[image]

azeey, 2024-04-08:
`expand_gz_topic_names` was added recently in https://github.com/gazebosim/ros_gz/pull/512 and ported to iron in https://github.com/gazebosim/ros_gz/pull/516. It has not been released yet (see https://github.com/gazebosim/ros_gz/blob/iron/ros_gz_bridge/CHANGELOG.rst). @ahcorde can you make a 0.254.1 release?

ahcorde, 2024-04-08:
I made releases for Humble and Iron
 - https://github.com/ros/rosdistro/pull/40515
 - https://github.com/ros/rosdistro/pull/40516

closed 2024-05-30 by azeey