gazebosim/ros_gz #536: Add custom GZ -> ROS mapping (Humble/Garden)

AntonioViscomi, 2024-04-13:
Hi all,
I need to use Gazebo Garden alongside ROS 2 Humble . I want to use the ros_gz_bridge to obtain measurements from virtual sensors on my robot and process them in ROS. <br>
In my ros_gz version ([this one](https://gazebosim.org/docs/harmonic/ros_installation#-gazebo-garden-with-ros-2-humble-iron-or-rolling-use-with-caution-) ) there is no mapping between <code>gz.msgs.NavSat</code> and <code>gps_msgs/msg/GPSFix</code>. <br>
I need this mapping because I want to transport also the ENU velocity in ROS, which is not possible using <code>sensor_msgs/msg/NavSatFix</code>

## Desired behavior
A map betweem <code>gz.msgs.NavSat</code> and <code>gps_msgs/msg/GPSFix</code> in ros_gz_bridge.

## Alternatives considered
I've cloned the "humble" branch (which should correspond to my installed version of ros_gz_bridge) and:

1.  I've included <code>gps_msgs.hpp</code> in <code> ros_gz/ ros_gz_bridge/include/ros_gz_bridge/convert</code>
2. Added <code>#include <ros_gz_bridge/convert/gps_msgs.hpp></code> to <code>convert.hpp</code>
3. Added <code>gps_msgs.cpp</code> in <code> ros_gz/ros_gz_bridge/src/convert </code>
4. Added inside  <code>ros_gz/ros_gz_bridge/mappings.py</code>: 
```python
  'gps_msgs': [
        Mapping('GPSFix', 'NavSat'),
    ], 
```

5. Added <code> <depend>gps_msgs</depend> </code> in <code>package.xml</code>

I am unsure if this will work because I've installed gps_msgs as a custom interface inside my custom interfaces package and I also need to ffigure out how to install this "new package" I've created.

### Additional context:
OS: Ubuntu 22.04.

AntonioViscomi, 2024-04-15:
Thanks a lot.
I've been able to solve this issue by myself, sharing the solution could be useful?

yschulz, 2024-04-26:
I would be more than happy to open another issue for this, but I think this fits here and since this issue is still open:

I am actually very interested in custom message mappings. With the ease of protobuf message generation, both in ros and in gazebo, it feels necessary that I should be able to generate my own custom mappings by configuration instead of programmatically. Even programmatically it seems necessary to fork this repo and inject the mappings directly like @AntonioViscomi  has suggested in his first comment. Is there any intent of adding a plugin like api or cmake generations or a cli tool to make this process easier?

azeey, 2024-04-26:
> Is there any intent of adding a plugin like api or cmake generations or a cli tool to make this process easier?

Yes, we would like to add this at some point, but I don't think there's an actual issue for it. I know for sure we've had discussions about it. The code was adapted from https://github.com/ros2/ros1_bridge/ which I've been told has a way to add custom messages, but I haven't tried it myself. If anyone would be willing to figure out how it's done for `ros1_bridge` and adapt it to `ros_gz_bridge`, it would be fantastic!

yschulz, 2024-04-26:
Thanks for the quick reply and the link, I can see the resemblance.

I would gladly take a look into it and see how this can be integrated here in the gz bridge. When I have a better overview I'd open a new issue!

AntonioViscomi, 2024-05-02:
Sorry for the late response, as long as I've seen there are more steps to do to get a custom <code>GZ -> ROS2</code> bridge. In the next few days I'll write what files need to be created and modified in addition to those in the original post. I need to reproduce all the steps to make sure  that the .hpp file names match with the ones in the .cpp files.