gazebosim/ros_gz #542: No way to get covariances out of GPS data

JayHerpin, 2024-05-01:
## Environment
* OS Version: Ubuntu 20.04 
* Source or binary build?
ros-iron-ros-gzharmonic-bridge/unknown,now 0.246.0-4jammy amd64 [installed]

Comparing the NavSatMsg messages from ros and gz, ros includes covariance information whereas the gz message does not. When bridging these messages from gz->ros, the covariance content is empty.

Contrast this with the IMU messages which include covariance in both cases. This makes integrating modern gazebo with something like robot localization under ros more difficult

```
std_msgs/Header header

# Satellite fix status information.
NavSatStatus status

# Latitude [degrees]. Positive is north of equator; negative is south.
float64 latitude

# Longitude [degrees]. Positive is east of prime meridian; negative is west.
float64 longitude

# Altitude [m]. Positive is above the WGS 84 ellipsoid
# (quiet NaN if no altitude is available).
float64 altitude

# Position covariance [m^2] defined relative to a tangential plane
# through the reported position. The components are East, North, and
# Up (ENU), in row-major order.
#
# Beware: this coordinate system exhibits singularities at the poles.
float64[9] position_covariance

# If the covariance of the fix is known, fill it in completely. If the
# GPS receiver provides the variance of each measurement, put them
# along the diagonal. If only Dilution of Precision is available,
# estimate an approximate covariance from that.

uint8 COVARIANCE_TYPE_UNKNOWN = 0
uint8 COVARIANCE_TYPE_APPROXIMATED = 1
uint8 COVARIANCE_TYPE_DIAGONAL_KNOWN = 2
uint8 COVARIANCE_TYPE_KNOWN = 3

uint8 position_covariance_type
```

```
## Description
* Expected behavior: <!-- Tell us what you expected to happen -->
* Actual behavior: <!-- What happened instead -->
syntax = "proto3";
package gz.msgs;
option java_package = "com.gz.msgs";
option java_outer_classname = "NavSatProtos";

/// \ingroup gz.msgs
/// \interface NavSat
/// \brief Data from a NavSat sensor
/// This replaces the GPS message.

import "gz/msgs/header.proto";

message NavSat
{
  /// \brief Optional header data
  Header header                = 1;

  /// \brief Latitude in degrees
  double latitude_deg          = 2;

  /// \brief Longitude in degrees
  double longitude_deg         = 3;

  /// \brief Altitude in meters
  double altitude              = 4;

  /// \brief East velocity in the ENU frame, in m / s
  double velocity_east         = 5;

  /// \brief North velocity in the ENU frame, in m / s
  double velocity_north        = 6;

  /// \brief Up velocity in the ENU frame, in m / s
  double velocity_up           = 7;

  /// \brief ID of reference frame
  string frame_id              = 8;
}

```

EmmanuelMess, 2025-02-12:
@mjcarroll Any news on this?

EmmanuelMess, 2025-03-03:
> [@EmmanuelMess](https://github.com/EmmanuelMess?rgh-link-date=2025-03-03T16%3A59%3A07.000Z) No news. I don't have much time for gazebo maintenance at the moment. I would be interested in reviewing contributions to address this, though.

I'm having trouble undertanding how the message passing system works, particularly, where the gazebo message is defined and its conversion to ROS 2 message is generated. Is there any documentation on this? The only thing I found is a [trigger](https://gazebosim.org/api/sim/9/triggeredpublisher.html) example, but I don't know if the GPS message works like that.