gazebosim/ros_gz #543: Build Failure on Ubuntu 22.04 with Protobuf Error

aadith-warrier, 2024-05-02:
## Environment
* OS Version: Ubuntu 22.04
* Source or binary build? Source build, humble branch 

## Description
* Expected behaviour: Successful build
* Actual behaviour: Build fails with (what I think is) a protobuf error

## Steps to reproduce

1. Create workspace
2. Update rosdep keys from https://github.com/osrf/osrf-rosdep, and install dependencies via rosdep
3. Run a colcon build

## Output

```
Starting >>> ros_gz_interfaces
Finished <<< ros_gz_interfaces [58.4s]                        
Starting >>> ros_gz_sim
Finished <<< ros_gz_sim [24.0s]                             
Starting >>> ros_gz_bridge
--- stderr: ros_gz_bridge                                          
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libprotobuf.a(arena.o): relocation R_X86_64_TPOFF32 against symbol `_ZN6google8protobuf8internal9ArenaImpl13thread_cache_E' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: failed to set dynamic section sizes: bad value
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/ros_gz_bridge.dir/build.make:936: libros_gz_bridge.so] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:183: CMakeFiles/ros_gz_bridge.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ros_gz_bridge [4min 49s, exited with code 2]

Summary: 2 packages finished [6min 12s]
  1 package failed: ros_gz_bridge
  1 package had stderr output: ros_gz_bridge
  10 packages not processed
```

azeey, 2024-05-02:
What version of protobuf do you have? I suspect you've installed a version other than the one supplied by your OS (Ubuntu 22.04).

aadith-warrier, 2024-05-04:
I have version 3.12.4, although I am not sure what version ships by default with 22.04

azeey, 2024-05-06:
Ubuntu 22.04 ships with 3.12.4, so that seems okay. What did you set `GZ_VERSION` to?

aadith-warrier, 2024-05-24:
harmonic

I assumed that ros_gz_bridge would have to be built from the source as there is no recommendation for integration between Humble and Harmonic on the ros2 branch. However, on the humble branch, it mentions that binaries are available. It would be nice if this is updated on the ros2 branch as well.

Sincere apologies for going dead for a while, I had my end sems.

azeey, 2024-05-24:
PRs are welcome regarding the documentation :) 

The fact that the error logs mention `/usr/lib/x86_64-linux-gnu/libprotobuf.a` is suspect. That's a static library and we should be linking against `libprotobuf.so`, the dynamic library. I was able to build the `humble` branch locally with `GZ_VERSION=harmonic` without any issues, so I'm not sure what could be wrong. Do you mind doing a clean build and seeing if you have the same error?

aadith-warrier, 2024-05-29:
I ran two clean builds, one on my system which failed with the same error and another on a docker container, which worked just fine. This definitely seems to be something specific to my system. Both are using the same protoc versions. 

Please feel free to close the issue unless you would like to investigate further.

azeey, 2024-05-29:
I haven't seen this issue come up before and like you said, it seems specific to your system, so I'll go ahead and close this. I hope you figure out what's causing it to try linking against a static library.

closed 2024-05-29 by azeey