gazebosim/ros_gz #547: executable name from gz_sim.launch.py

christophfroehlich, 2024-05-10:
How is this supposed to work with ruby?

https://github.com/gazebosim/ros_gz/blob/78dc4823121f085594e6028a93f1e571eb04f58b/ros_gz_sim/launch/gz_sim.launch.py.in#L45-L51

The executable name just remains `ruby $(which gz) sim`, see for example the [CI in gz_ros2_control](https://github.com/ros-controls/gz_ros2_control/actions/runs/8991838025/job/24700286964#step:6:476). Is there something missing to be installed?

mjcarroll, 2024-05-14:
I believe that it's just using simple shell expansion `$()` which gets evaluated before running `ruby`.

christophfroehlich, 2024-05-14:
Ok, so might it be that this an effect of `launch.actions.ExecuteProcess`, which creates a logger with a name being this command, but the command gets evaluated correctly when actually launching the executable?
At least, it does not make the log files very readable (neither in our CI, nor in log files posted e.g. on stack exchange)

christophfroehlich, 2024-05-14:
May I ask why there is a need for ruby at all?

azeey, 2024-05-14:
#500 and related PRs should help with this since Gazebo will be ran as a ROS node.

> May I ask why there is a need for ruby at all?

Ruby is needed because the `gz` is written in Ruby (https://github.com/gazebosim/gz-tools/blob/gz-tools2/src/gz.in). It's the tool that aggregates all of the commands available from all the Gazebo libraries. There's an effort to use binary executables (https://github.com/gazebosim/gz-tools/issues/7) for this, but we have a ways to go.

christophfroehlich, 2024-05-14:
Ok, thanks for the explanation.

closed 2024-05-14 by christophfroehlich