gazebosim/ros_gz #549: RosGzPointCloud Not Loading

BobbieZain, 2024-05-20:
## Environment
* OS Version: Ubuntu 22.04
* Binary build: 2.5.5-1jammy.20240404.154053

## Description
* Expected behavior: The depth image can be seen in image display
* Actual behavior: The image display cannot be dropped down and the plugin is not correctly loaded

## Steps to reproduce

1.Go to [ros_gz_point_cloud/examples](https://github.com/gazebosim/ros_gz/tree/78dc4823121f085594e6028a93f1e571eb04f58b/ros_gz_point_cloud/examples)
2.Run `ign gazebo -r examples/depth_camera.sdf`

## Output
Error message:
`[Err] [SystemLoader.cc:94] Failed to load system plugin [RosGzPointCloud] : couldn't find shared library.
`
Screenshot:
[image]

BobbieZain, 2024-05-22:
Hello, do I need to install the `RosGzPointCloud` library? If so, how?

azeey, 2024-05-22:
The `PointCloud` plugin is unfortunately in a broken state right now. There's a [COLCON_IGNORE](https://github.com/gazebosim/ros_gz/blob/humble/ros_gz_point_cloud/COLCON_IGNORE) file in the package.

A workaround for this is to use `ros_gz_bridge` to bridge point cloud messages from Gazebo to ROS. [Here](https://github.com/gazebosim/ros_gz/blob/humble/ros_gz_sim_demos/launch/rgbd_camera_bridge.launch.py) is an example.

DanielBar98, 2024-11-01:
Are there any updates on this?

caguero, 2024-11-01:
We're looking into an alternative, stay tuned for news soon,

SuperJappie08, 2024-11-29:
I attempted to get this working on ROS Jazzy with Gazebo Harmonic, which I also mentioned in #360.

Here is the same comment, since it might also apply:
> I attempted to port this for ROS Jazzy and Gazebo Harmonic. (Because it looked useful and doable, and I wanted more about Gazebo)
>
> However, even after converting it to a rendering plugin, the performance improvements gazebosim/gz-sensors#447 and gazebosim/gz-sensors#413, it is no longer possible to get data via the renderer directly without subscribing to the (gazebo) topic.
> 
> (My branch lives here, although not functioning as expected [SuperJappie08/ros_gz jazzy-point-cloud](https://github.com/SuperJappie08/ros_gz/tree/jazzy-point-cloud))

SyedMIrtazaHyder, 2025-04-13:
How is the example in ros_gz_bridge_demos working yet not the plugin itself?
> A workaround for this is to use `ros_gz_bridge` to bridge point cloud messages from Gazebo to ROS. [Here](https://github.com/gazebosim/ros_gz/blob/humble/ros_gz_sim_demos/launch/rgbd_camera_bridge.launch.py) is an example.

How can this workaround be extended for other projects?