gazebosim/ros_gz #560: Gz Sim does not exist

rfpg1, 2024-06-08:
## Operating System
OS Version: Ubuntu 24.04

## Ros Version 
ROS 2 Jazzy (Debian Packages)

## Gazebo Version
Binary install: Gazebo Harmonic LTS

## Description
When installing the ros-jazzy-ros-gz-bridge gz sim is not recognized as command

## Steps to reproduce issue
1. Normal installation of the guide: https://docs.ros.org/en/jazzy/Installation/Ubuntu-Install-Debians.html

2. Normal installation of the guide: https://gazebosim.org/docs/harmonic/install_ubuntu

3. sudo apt install ros-jazzy-ros-gz-bridge

## Output
Output of gz:

```
The 'gz' command provides a command line interface to the Gazebo Tools.

  gz <command> [options]

List of available commands:

  help:          Print this help text.
  log:           Record or playback topics.
  msg:           Print information about messages.
  param:         List, get or set parameters.
  service:       Print information about services.
  topic:         Print information about topics.

Options:

  --force-version <VERSION>  Use a specific library version.
  --versions                 Show the available versions.
  --commands                 Show the available commands.
Use 'gz help <command>' to print help for a command.
```

Also opened an issue on: https://github.com/gazebosim/gz-sim/issues/2438

But realized here would make more sense

Any ideas to fix? Tried uninstalling but did not work and also would not be a solution since I need the bridge

rfpg1, 2024-06-10:
Managed to fix this

All I had to do was: ros-jazzy-gz-sim-vendor as mencioned here: https://community.gazebosim.org/t/gazebo-harmonic-packages-available-on-ubuntu-noble/2772/8

closed 2024-06-10 by rfpg1

Kvitravne, 2025-11-24:
I had the same issue. I think it’s a conflict between ROS2 and Gazebo, so I removed both. Then I installed Gazebo first, and ROS2 afterward. That fixed the issue for me.

azeey, 2025-11-24:
Please follow https://gazebosim.org/docs/harmonic/ros_installation/ when using Gazebo with ROS. In your case, you'll need to install `ros-jazzy-ros-gz` and re-source `/opt/jazzy/setup.bash` for Gazebo to work properly.