gazebosim/ros_gz #561: What Version of Protobuf are we using?

woensug-choi, 2024-06-18:
## Environment
* OS Version: MacOS Sonoma, Applesilicon
* Latest ROS2 Source Code building
  
  
 

## Description
* Expected behavior: Compile success
* Actual behavior: 

## Steps to reproduce

- Mac Sonoma with Jazzy ROS2 (source code) and Gazebo Garden (using brew, of course with gz-msg, gz-transport etc along with it with brew)

## Output
```bash
In file included from /Users/woensug/ros_gz_ws/src/ros_gz_for_mac/ros_gz_sim/test/test_create.cpp:17:
In file included from /opt/homebrew/include/gz/msgs10/gz/msgs/entity_factory.pb.h:12:
In file included from /opt/homebrew/include/gz/msgs10/gz/msgs/details/entity_factory.pb.h:32:
In file included from /opt/homebrew/include/gz/msgs10/gz/msgs/light.pb.h:12:
/opt/homebrew/include/gz/msgs10/gz/msgs/details/light.pb.h:16:2: error: "Protobuf C++ gencode is built with an incompatible version of"
#error "Protobuf C++ gencode is built with an incompatible version of"
 ^
/opt/homebrew/include/gz/msgs10/gz/msgs/details/light.pb.h:17:2: error: "Protobuf C++ headers/runtime. See"
#error "Protobuf C++ headers/runtime. See"
 ^
/opt/homebrew/include/gz/msgs10/gz/msgs/details/light.pb.h:18:2: error: "https://protobuf.dev/support/cross-version-runtime-guarantee/#cpp"
#error "https://protobuf.dev/support/cross-version-runtime-guarantee/#cpp"
 ^
In file included from /Users/woensug/ros_gz_ws/src/ros_gz_for_mac/ros_gz_sim/test/test_create.cpp:17:
In file included from /opt/homebrew/include/gz/msgs10/gz/msgs/entity_factory.pb.h:12:
In file included from /opt/homebrew/include/gz/msgs10/gz/msgs/details/entity_factory.pb.h:32:
In file included from /opt/homebrew/include/gz/msgs10/gz/msgs/light.pb.h:12:
In file included from /opt/homebrew/include/gz/msgs10/gz/msgs/details/light.pb.h:33:
In file included from /opt/homebrew/include/gz/msgs10/gz/msgs/pose.pb.h:12:
/opt/homebrew/include/gz/msgs10/gz/msgs/details/pose.pb.h:16:2: error: "Protobuf C++ gencode is built with an incompatible version of"
#error "Protobuf C++ gencode is built with an incompatible version of"
 ^
/opt/homebrew/include/gz/msgs10/gz/msgs/details/pose.pb.h:17:2: error: "Protobuf C++ headers/runtime. See"
#error "Protobuf C++ headers/runtime. See"
 ^
/opt/homebrew/include/gz/msgs10/gz/msgs/details/pose.pb.h:18:2: error: "https://protobuf.dev/support/cross-version-runtime-guarantee/#cpp"
#error "https://protobuf.dev/support/cross-version-runtime-guarantee/#cpp"
 ^
In file included from /Users/woensug/ros_gz_ws/src/ros_gz_for_mac/ros_gz_sim/test/test_create.cpp:17:
In file included from /opt/homebrew/include/gz/msgs10/gz/msgs/entity_factory.pb.h:12:
In file included from /opt/homebrew/include/gz/msgs10/gz/msgs/details/entity_factory.pb.h:32:
In file included from /opt/homebrew/include/gz/msgs10/gz/msgs/light.pb.h:12:
In file included from /opt/homebrew/include/gz/msgs10/gz/msgs/details/light.pb.h:33:
In file included from /opt/homebrew/include/gz/msgs10/gz/msgs/pose.pb.h:12:
In file included from /opt/homebrew/include/gz/msgs10/gz/msgs/details/pose.pb.h:31:
In file included from /opt/homebrew/include/gz/msgs10/gz/msgs/vector3d.pb.h:12:
/opt/homebrew/include/gz/msgs10/gz/msgs/details/vector3d.pb.h:16:2: error: "Protobuf C++ gencode is built with an incompatible version of"
#error "Protobuf C++ gencode is built with an incompatible version of"
 ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
gmake[2]: *** [CMakeFiles/test_create.dir/build.make:76: CMakeFiles/test_create.dir/test/test_create.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:381: CMakeFiles/test_create.dir/all] Error 2
20 errors generated.
gmake[2]: *** [CMakeFiles/create.dir/build.make:76: CMakeFiles/create.dir/src/create.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:248: CMakeFiles/create.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ros_gz_sim [9.45s, exited with code 2]
Aborted  <<< ros_gz_interfaces [14.6s] 
```

woensug-choi, 2024-06-18:
Never mind. I bypassed the problem by source compiling the `Gazebo Harmonic` and use the packages being compiled in the way.

https://github.com/IOES-Lab/ros_gz_for_mac

closed 2024-06-18 by woensug-choi

azeey, 2024-06-18:
We don't pin the protobuf version in the homebrew formula (see https://github.com/osrf/homebrew-simulation/blob/5f50a0c8a72624157fd77357f8803a4b229636b1/Formula/gz-msgs10.rb#L23), but it might be that a new protobuf was released after we built the bottle. The bottles were rebuilt [yesterday](https://github.com/osrf/homebrew-simulation/pull/2677), so maybe you need to reinstall. Building Gazebo from source like you've done would also work.