gazebosim/ros_gz #580: Not able to build ros_gz_bridge

Luigip98, 2024-07-23:
ROS2 version : humble

ros_gz/ros_gz_bridge/include/ros_gz_bridge/convert/ros_gz_interfaces.hpp:39:10: fatal error: ros_gz_interfaces/msg/entity_wrench.hpp: No such file or directory
   39 | #include <ros_gz_interfaces/msg/entity_wrench.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/ros_gz_bridge.dir/build.make:251: CMakeFiles/ros_gz_bridge.dir/src/service_factories/ros_gz_interfaces.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:183: CMakeFiles/ros_gz_bridge.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ros_gz_bridge [26.6s, exited with code 2]

How can I avoid this?

mjcarroll, 2024-07-23:
Have you also rebuild the `ros_gz_interfaces` package?  It appears there is a new message file that is needed that you don't have generated yet.

Luigip98, 2024-07-23:
I've cloned the whole ros_gz repo (humble branch) and I have no other building errors.

imcelroy, 2024-10-22:
Is there any progress on this issue? I have the same problem when building from source on the `humble` branch.