gazebosim/ros_gz #581: ros_gz_bridge: GZ -> ROS /clock publisher

asherikov, 2024-07-23:
Hello, I've been investigating poor performance of the /clock topic and noticed a couple of issues:
1. Publisher does not use ClockQoS by default (https://github.com/ros2/rclcpp/blob/rolling/rclcpp/include/rclcpp/qos.hpp#L364), which would be beneficial.
2. Heavy sensor topics (cameras, lidars) interfere with the clock publisher by introducing delays since they share the same single threaded spinner.

The issue can be addressed with a dedicated bridge instance with qos and node name overrides, but given the importance of the clock topic it makes sense to create a dedicated node to simplify usage of the bridge.

azeey, 2024-07-29:
It is possible to have a separate bridge (I.e, a separate process) for the clock node with its own QOS setting, but having better defaults would be great. @asherikov are you interested in working on this?

asherikov, 2024-07-30:
Hi, it looks like adding a dedicated node requires the least amount of work, otherwise the bridge would require significant refactoring: custom settings for a specific topic, multithreaded spinning, etc. I have solved my problems with a dedicated instance of the bridge, so I am currently not interested in working on a better solution.

azeey, 2024-07-30:
Thanks for the response! I'll close this issue then. I have opened another [issue](https://github.com/gazebosim/ros_gz/issues/583) to better document QoS settings.

closed 2024-07-30 by azeey