gazebosim/ros_gz #582: I would like to use EntityWrench

yama3-8888, 2024-07-24:
## Environment
Ubuntu 22.04.4 LTS
* Source, branch is ros2
* ROS2: Humble
* Ignition Gazebo: Fortress

## Description
* Expected behavior: The communication connection between ROS 2 and Ignition Gazebo is success via ros_gz_bridge.
* Actual behavior: Failed to create a bridge for topic with ROS2 type to topic with Gazebo Transport type.

## Steps to reproduce
1. Enter "apt install ros-humble-ros-gz" into the command line.
2. Followed the step Readme by ros2 branch  (https://github.com/gazebosim/ros_gz?tab=readme-ov-file)
3. Enter "ros2 run ros_gz_bridge parameter bridge /world/simu/wrench@ros_gz_interfaces/msg/EntityWrench@gz.msgs.EntityWrench" into the command line.

## Output
[INFO] [1721819972.818906666] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/world/simu/wrench (gz.msgs.EntityWrench) -> /world/simu/wrench (ros_gz_interfaces/msg/EntityWrench)] (Lazy 0)
[WARN] [1721819972.819089812] [ros_gz_bridge]: Failed to create a bridge for topic [/world/simu/wrench] with ROS2 type [ros_gz_interfaces/msg/EntityWrench] to topic [/world/simu/wrench] with Gazebo Transport type [gz.msgs.EntityWrench]

ahcorde, 2024-07-24:
`ros-humble-ros-gz` is still not sync in the debian packages I made the release 2 days ago. It should happen in the following weeks, for now you can compile the code using the `humble` branch.

yama3-8888, 2024-07-25:
Thank you for the rapid response.
## Discrioption
I changed the branch to `humble` and followed the same procedure, but I encountered a problem where the `colcon build` would freeze.

## Enviroment
Source, branch is latest `humble`

### Steps to reproduce
1. I followed the step Readme by `humble` branch.
2. I executed build with `colcon build ` into my workspace.
### Result
It freezes during `colcon build`.

### Package suspected to be the cause
1.  I executed build with `colcon build --packages-select ros_gz_bridge` into my workspace.

### Result
Similarly, It freezes during `colcon build --packages-select ros_gz_bridge`.

yama3-8888, 2024-07-25:
Thank you for cooporation!
## Discription
 The communication connection between ROS 2 and Ignition Gazebo is success via ros_gz_bridge.

## What is problem
The problem was with my computer specs.

## approach
When I ran `colcon build --parallel-workers=1 --executor sequential` on a high spec computer, the build went through without freezing.

## Input into my commandline
`ros2 run ros_gz_bridge parameter_bridge /world/simu/wrench@ros_gz_interfaces/msg/EntityWrench@gz.msgs.EntityWrench`

## Out put
```
[INFO] [1721891599.825770895] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/world/simu/wrench (gz.msgs.EntityWrench) -> /world/simu/wrench (ros_gz_interfaces/msg/EntityWrench)] (Lazy 0)
[INFO] [1721891599.829781391] [ros_gz_bridge]: Creating ROS->GZ Bridge: [/world/simu/wrench (ros_gz_interfaces/msg/EntityWrench) -> /world/simu/wrench (gz.msgs.EntityWrench)] (Lazy 0)
```

ahcorde, 2024-07-25:
Closing this issue then, Feel free to reopen it if you have more problems

closed 2024-07-25 by ahcorde