gazebosim/ros_gz #589: Issue with ros2 jazzy and gazebo harmonic ros integration

treize06, 2024-08-19:
Hello, I’m quite new wit ROS and gazebo and I have been having an issue for several days:
when following the tutorial : [Spawn a Gazebo model from ROS 2 — Gazebo harmonic documentation](https://gazebosim.org/docs/harmonic/ros2_spawn_model/)

It doesn’t work, after trying many time to reinstall, I found out that I don’t have the needed files in /opt/ros/jazzy/share/ros_gz_sim/launch

looking at the github it seems the jazzy/harmonic branch doesn’t have those files ( [ros_gz/ros_gz_sim/launch at jazzy · gazebosim/ros_gz · GitHub](https://github.com/gazebosim/ros_gz/tree/jazzy/ros_gz_sim/launch) )

What should I do to be able to make it work ?

thanks

azeey, 2024-08-19:
Yeah, we need to backport those changes to Jazzy. In the meantime, we should probably update the documentation to say that this requires `rolling`.

cc @caguero

Amronos, 2024-10-20:
I think this issue can be closed now.