gazebosim/ros_gz #591: Bridging same topic via multiple nodes leads to a conflict

k12onoss, 2024-08-29:
## Environment
* OS Version:  Ubuntu 24.04
* Source or binary build?
  
  
  binary

## Description
So, I ran into this issue while working with multi-robot simulation. I am using the [`cloned_multi_tb3_simulation_launch.py`](https://github.com/ros-navigation/navigation2/blob/main/nav2_bringup/launch/cloned_multi_tb3_simulation_launch.py) from the nav2 package, which launches a separate bridge for each instance of the tb3 robot that you spawn. As a result, the `/clock` topic is bridged multiple times. This leads to the following issues:
- `Detected jump back in time. Clearing TF buffer.`
- `Message Filter dropping message: frame 'base_scan' at time 7.600 for reason 'the timestamp on the message is earlier than all the data in the transform cache'`
* Expected behavior: 
There's no mismatch in the messages published.
* Actual behavior: 
Leads to some conflict. I've verified this by essentially omitting the `/clock` topic from being bridged for each robot instance and instead have a separate bridge for it.

## Steps to reproduce

1. Launch the `cloned_multi_tb3_simulation_launch.py` file.

azeey, 2024-08-29:
I'm not sure what the fix should be here. It seems to me the right thing to do is to only bridge global topics like `/clock` only once.

azeey, 2024-11-11:
I'm going to close this issue since it seems like a configuration problem. I think the fix is to make sure the `/clock` topic is briged once in your project or, in this case, in `nav2_bringup`.

closed 2024-11-11 by azeey