gazebosim/ros_gz #608: Add deadline and liveliness to the qos_overriding_options

norakon, 2024-09-13:
Hello,

I would like to request a feature enhancement for the ros_gz_bridge package. Currently, the package allows configuring the following Quality of Service (QoS) settings via parameters: Depth, Durability, History, and Reliability. This is very useful, but it would be even more beneficial if we could also configure the *Deadline* and *Liveliness* properties of QoS.

## Feature Request:

Please add possibility to configure the QoS Deadline and Liveliness properties via the ```qos_overrides``` parameter.

## Context and Use Case:

Configuring Deadline and Liveliness properties can be critical for applications with stringent real-time requirements or for ensuring reliable message delivery and processing in complex systems. Adding these parameters will enhance the flexibility of the ros_gz_bridge package, making it more adaptable to diverse scenarios and requirements.

## Reference:

The currently allowed configurable properties can be seen in the qos_overriding_options in  [factory.hpp](https://github.com/gazebosim/ros_gz/blob/f2f0cedf6fb7bcf959d6db0f44bd1a08c203741e/ros_gz_bridge/src/factory.hpp#L66). I believe extending this to include Deadline and Liveliness would be a valuable addition.

norakon, 2024-09-13:
I took the liberty to add the missing qos policies here: https://github.com/norakon/ros_gz/compare/ros2...extend_qos_overriding_options

ahcorde, 2024-09-13:
Do you mind to open a PR ?

closed 2024-09-19 by norakon