gazebosim/ros_gz #614: `ros-humble-ros-gzharmonic` binary for ARM64 architecture

quantumxt, 2024-09-18:
Hi there, just wondering if there's a binary for `arm64` architecture? 

I compared the following Packages list:
- `amd64` (https://packages.osrfoundation.org/gazebo/ubuntu-stable/dists/jammy/main/binary-amd64/) 
- `arm64` (https://packages.osrfoundation.org/gazebo/ubuntu-stable/dists/jammy/main/binary-arm64/)

`ros-humble-ros-gzharmonic` was only listed inside `amd64`, but not `arm64`.

martincerven, 2024-10-04:
Hmm and I thought `E: Unable to locate package ros-humble-ros-gzharmonic` is due to wrong documentation  😓

azeey, 2024-11-05:
Unfortunately, we do not provide binaries for ARM64 and there is currently no plan to do so. ROS Jazzy does have a version of Gazebo and ros_gz in ARM64, so I suggest upgrading.