gazebosim/ros_gz #615: How to use MaterialColor Msg

yanmiao2, 2024-09-23:
Hi, I'm very new to gazebo and ros. I'd like to change material color while launching the simulator, potential using ros? or at least gazebo command line. 

I see that it's possible by #486 , but i did not see any tutorials on how to make it happen. I can build the bridge using `ros2 run ros_gz_bridge parameter_bridge /world/material_color/material_color@ros_gz_interfaces/msg/MaterialColor@gz.msgs.MaterialColor`, but there's nothing being published to that topic, did I miss a plug-in? Can someone walk me through what i should do? Thanks

I'm using a very simple sdf from https://github.com/gazebosim/gz-sim/pull/2286/files#diff-e3c9cb7abd311a035b602cba1847f3d96c1d7e9cfe0e58c863a8355551700738
```
<?xml version="1.0" ?>
<sdf version="1.10">
  <world name="material_color">
    <physics name="1ms" type="ode">
      <max_step_size>0.001</max_step_size>
      <real_time_factor>0</real_time_factor>
    </physics>
    <plugin
      filename="gz-sim-scene-broadcaster-system"
      name="gz::sim::systems::SceneBroadcaster">
    </plugin>
    <plugin
      filename="gz-sim-user-commands-system"
      name="gz::sim::systems::UserCommands">
    </plugin>
    <plugin
      filename="gz-sim-sensors-system"
      name="gz::sim::systems::Sensors">
      <render_engine>ogre2</render_engine>
    </plugin>

    <model name="sphere_0">
      <pose>0 0.0 0.0 0 0 0</pose>
      <link name="sphere_link_0">
        <!-- Added a render sensor to trigger RenderUtil:Update -->
        <sensor name="camera" type="camera">
          <pose>1 0 1.3 0 0 0</pose>
          <camera>
            <horizontal_fov>1.047</horizontal_fov>
            <image>
              <width>320</width>
              <height>240</height>
            </image>
            <clip>
              <near>0.1</near>
              <far>100</far>
            </clip>
          </camera>
          <always_on>1</always_on>
          <update_rate>30</update_rate>
          <visualize>false</visualize>
          <topic>camera</topic>
        </sensor>
        <visual name="sphere_visual">
          <geometry>
            <sphere>
              <radius>0.5</radius>
            </sphere>
          </geometry>
          <material>
            <ambient>0.3 0.3 0.3 1</ambient>
            <diffuse>0.3 0.3 0.3 1</diffuse>
            <specular>0.3 0.3 0.3 1</specular>
          </material>
        </visual>
      </link>
    </model>
    <model name="sphere_1">
      <pose>0.5 1.0 0.0 0 0 0</pose>
      <link name="sphere_link_1">
        <visual name="sphere_visual">
          <geometry>
            <sphere>
              <radius>0.5</radius>
            </sphere>
          </geometry>
          <material>
            <ambient>0.3 0.3 0.3 1</ambient>
            <diffuse>0.3 0.3 0.3 1</diffuse>
            <specular>0.3 0.3 0.3 1</specular>
          </material>
        </visual>
      </link>
    </model>
  </world>
</sdf>
```

yanmiao2, 2024-09-24:
Oh nvm, I figured it out, basically the rostopic is not to display current material color, but used to be written to update new color.

Basically the procedure like following:
1. Launch the bridge with topic `ros2 run ros_gz_bridge parameter_bridge /world/material_color/material_color@ros_gz_interfaces/msg/MaterialColor@gz.msgs.MaterialColor`
2. Write a simple ros publisher node that write to `/world/material_color/material_color` with type `gz.msgs.MaterialColor`

bperseghetti, 2024-09-24:
If doing it from a basic node it can be as easy as running:

```bash
gz sim default.sdf
```
Now add a box to the world from the shape menu:
[image]

Notice it has a `box_visual`
[image]

Start a bridge:
```bash
ros2 run ros_gz_bridge parameter_bridge /world/default/material_color@ros_gz_interfaces/msg/MaterialColor@gz.msgs.MaterialColor
```
Run a node to change the material color (in python for easiest readability):

```python
import rclpy
import numpy as np
from rclpy.node import Node
from ros_gz_interfaces.msg import MaterialColor

class MaterialTester(Node):
    def __init__(self):
        super().__init__('material_tester_node')
        self.MaterialColorTopic = '/world/default/material_color'
        self.MaterialColorPub = self.create_publisher(MaterialColor,'{:s}'.format(self.MaterialColorTopic), 1)
        timer_period = 0.02
        self.i=0
        self.timer = self.create_timer(timer_period, self.timerCallback)
     
    def timerCallback(self):
        if self.i < 49:
            self.i+=1
        elif self.i >= 49:
            self.i=0
        msgM=MaterialColor()
        msgM.entity_match=1
        msgM.entity.name="box_visual"
        msgM.diffuse.b = 1.0
        msgM.diffuse.g = .1-.1*np.sin(2*np.pi*self.i/49+np.pi/2)
        msgM.diffuse.r = .1-.1*np.sin(2*np.pi*self.i/49+np.pi/2)
        msgM.diffuse.a = 1.0*np.sin(2*np.pi*self.i/49)
        msgM.emissive=msgM.specular=msgM.ambient=msgM.diffuse
        self.MaterialColorPub.publish(msgM)
        return
if __name__ == '__main__':
    rclpy.init()
    MT = MaterialTester()
    rclpy.spin(MT)
    MT.destroy_node()
    rclpy.shutdown()
```

Watch colors change:
[Screencast from 2024-09-24 15-51-33.webm](https://github.com/user-attachments/assets/8ea727ad-7b93-4cd1-97ae-a3c8aee994d6)

azeey, 2024-11-07:
Closing since this doesn't appear to be a bug and the question seems to have been addressed.

closed 2024-11-07 by azeey