gazebosim/ros_gz #619: The IMU data interval is unstable with high update frequency.

Rubbishwang, 2024-10-17:
I used Gazebo 9.19.0, ros noetic, mavros, mavlink and px4 in Ubuntu20.04 environment for co-simulation. Run the [rosservice call /mavros/set_message_interval 31 100 ](url) command to set the IMU update frequency at 100 Hz. I expect the IMU to refresh at a constant 10ms interval, but in practice it always fluctuates between 8-12ms. I tried to modify [max_step_size](url) and [real_time_update_rate](url) in the world file, but the problem persisted.What can I do to keep the IMU update rate as stable as possible at 10ms?
Part of my world file is as follows:
  <physics name='default_physics' default='0' type='ode'>
  <ode>
    <solver>
      <type>quick</type>
      <iters>200</iters>
      <sor>1.3</sor>
      <use_dynamic_moi_rescaling>0</use_dynamic_moi_rescaling>
    </solver>
    <constraints>
      <cfm>0</cfm>
      <erp>0.2</erp>
      <contact_max_correcting_vel>100</contact_max_correcting_vel>
      <contact_surface_layer>0.001</contact_surface_layer>
    </constraints>
  </ode>
  <max_step_size>0.0001</max_step_size>
  <real_time_factor>1</real_time_factor>
  <real_time_update_rate>10000</real_time_update_rate>
</physics>

<plugin name='rotors_gazebo_imu_plugin' filename='libgazebo_imu_plugin.so'>
    <robotNamespace/>
    <linkName>iris::/imu_link</linkName>
    <imuTopic>/imu</imuTopic>
    <gyroscopeNoiseDensity>0.0</gyroscopeNoiseDensity>
    <gyroscopeRandomWalk>0</gyroscopeRandomWalk>
    <gyroscopeBiasCorrelationTime>1000000.0</gyroscopeBiasCorrelationTime>
    <gyroscopeTurnOnBiasSigma>0.0</gyroscopeTurnOnBiasSigma>
    <accelerometerNoiseDensity>0.0</accelerometerNoiseDensity>
    <accelerometerRandomWalk>0.0</accelerometerRandomWalk>
    <accelerometerBiasCorrelationTime>1000000.0</accelerometerBiasCorrelationTime>
    <accelerometerTurnOnBiasSigma>0.0</accelerometerTurnOnBiasSigma>
  </plugin>
I have been puzzled by this question for a long time. I hope for you answer. Thank you!

azeey, 2024-10-17:
This seems like a question better suited for robotics.stackexchange.com. I encourage you to post your question there.
Also, it looks like you're using [Gazebo-classic](https://gazebosim.org/about), but this repository is for the new Gazebo. It would help if you make the distinction when you ask the question by putting the `gazebo-classic` tag.

closed 2024-10-17 by azeey

Rubbishwang, 2024-10-19:
> This seems like a question better suited for robotics.stackexchange.com. I encourage you to post your question there. Also, it looks like you're using [Gazebo-classic](https://gazebosim.org/about), [...]

Thanks for you reply! I will try to post my question on robotics.stackexchange.com.