gazebosim/ros_gz #626: libEGL warning: NEEDS EXTENSION: falling back to kms_swrast terminate called after throwing an instance of 'Ogre::InvalidParametersException'

FaterYU, 2024-10-27:
## Environment
* OS Version: Ubuntu 22.04
* Source: Ignition Fortress
* ROS2  version: Humble

## Steps to reproduce

1. Open one terminal run
```
ign gazebo empty.sdf
```
2. Open another terminal run 
```
ign service -s /world/empty/create --reqtype ignition.msgs.EntityFactory --reptype ignition.msgs.Boolean --timeout 1000 --req 'sdf_filename: "/ros_ws/src/mobot_description/urdf/mobot.urdf", name: "mobot_description"'
```

I make sure the mesh path is correct and model is works in `rviz2` correctly.

I try to delete the `visual` part in urdf and then it does not raise the errors, but I need the visual part.

The `mobot.urdf` is:
```
<?xml version="1.0" ?>
<!-- =================================================================================== -->
<!-- |    This document was autogenerated by xacro from mobot_overall.urdf.xacro       | -->
<!-- |    EDITING THIS FILE BY HAND IS NOT RECOMMENDED                                 | -->
<!-- =================================================================================== -->
<robot name="mobot_description">
  <link name="base_link">
    <inertial>
      <origin rpy="0 0 0" xyz="0.0105481626504806 0.000121730205140724 -0.0536615390649162"/>
      <mass value="11.9970840115073"/>
      <inertia ixx="0.0265731409814962" ixy="1.52214545840681E-05" ixz="-0.000581289129660373" iyy="0.0533099030677871" iyz="-5.74575531055247E-07" izz="0.0503147943878383"/>
    </inertial>
    <!-- <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="/ros_ws/src/mobot_description/meshes/base_link.STL"/>
      </geometry>
      <material name="">
        <color rgba="1 1 1 1"/>
      </material>
    </visual> -->
    <collision>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="/ros_ws/src/mobot_description/meshes/base_link.STL"/>
      </geometry>
    </collision>
  </link>
</robot>
```

## Output

```
➜  /ros_ws ign gazebo empty.sdf
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
libEGL warning: MESA-LOADER: failed to open vgem: /usr/lib/dri/vgem_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)

libEGL warning: NEEDS EXTENSION: falling back to kms_swrast
terminate called after throwing an instance of 'Ogre::InvalidParametersException'
  what():  OGRE EXCEPTION(2:InvalidParametersException): StagingBuffer cannot map 0 bytes in StagingBuffer::mapChecks at /build/ogre-next-UFfg83/ogre-next-2.2.5+dfsg3/OgreMain/src/Vao/OgreStagingBuffer.cpp (line 64)
Stack trace (most recent call last) in thread 230276:
#10   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in
#9    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f708b5f584f, in
#8    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f708b563ac2, in
#7    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f7085c7e99d, in
#6    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f7085c7cf90, in qTerminate()
#5    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f7087aff276, in std::terminate()
#4    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f7087aff20b, in
#3    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f7087af3b9d, in
#2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f708b4f77f2, in abort
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f708b511475, in raise
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f708b5659fc, in pthread_kill
Aborted (Signal sent by tkill() 230206 0)
```

iche033, 2024-10-29:
looks like it's an mesa driver issue? https://forums.developer.nvidia.com/t/vgem-dri-so-file-is-missing-libegl-warning/244383/7

You can try running with ogre 1.x render engine, e.g. `ign gazebo --render-engine ogre empty.sdf`

pep248, 2024-11-07:
> looks like it's an mesa driver issue? https://forums.developer.nvidia.com/t/vgem-dri-so-file-is-missing-libegl-warning/244383/7
> 
> You can try running with ogre 1.x render engine, e.g. `ign gazebo --render-engine ogre empty.sdf`

How can I force to always use ogre 1.x?

iche033, 2024-11-07:
> How can I force to always use ogre 1.x?

easiest way is to do it through the cmd line arg `--render-engine ogre`: this sets both the GUI window and the camera sensors to render using `ogre`. You can also edit `~/.gz/sim/<version>/gui.config` and update `<engine>` to `ogre` for the GUI window.

azeey, 2024-12-16:
@pep248 , @iche033 can this issue be closed?

iche033, 2024-12-16:
yes, we can close this.

closed 2024-12-16 by iche033

pep248, 2024-12-17:
Yes.