gazebosim/ros_gz #630: gzserver Does not Find Some Gazebo Plugins

Amronos, 2024-11-02:
## Environment
* OS Version: Ubuntu 24.04
* Source or binary build?
 Source - ad225a2

## Description
* Expected behavior: When launching gazebo through the [gz_server.launch.py](https://github.com/gazebosim/ros_gz/blob/ros2/ros_gz_sim/launch/gz_server.launch.py) launch file, I expected gazebo to find the `gz_ros2_control-system` plugin and not report an error.
* Actual behavior: Gazebo does not find the plugin and outputs ``Failed to load system plugin [gz_ros2_control-system] : Could not find shared library.``

More information can be found in https://github.com/ros-controls/gz_ros2_control/issues/390

## Steps to reproduce
Use the following launch file (A slightly modified version of [diff_drive_example.launch.py](https://github.com/ros-controls/gz_ros2_control/blob/rolling/gz_ros2_control_demos/launch/diff_drive_example.launch.py) that uses [gz_server.launch.py](https://github.com/gazebosim/ros_gz/blob/ros2/ros_gz_sim/launch/gz_server.launch.py)):

```python
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument, IncludeLaunchDescription
from launch.actions import RegisterEventHandler
from launch.event_handlers import OnProcessExit
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.substitutions import Command, FindExecutable, LaunchConfiguration, PathJoinSubstitution

from launch_ros.actions import Node
from launch_ros.substitutions import FindPackageShare

def generate_launch_description():
    # Launch Arguments
    use_sim_time = LaunchConfiguration('use_sim_time', default=True)

    # Get URDF via xacro
    robot_description_content = Command(
        [
            PathJoinSubstitution([FindExecutable(name='xacro')]),
            ' ',
            PathJoinSubstitution(
                [FindPackageShare('gz_ros2_control_demos'),
                 'urdf', 'test_diff_drive.xacro.urdf']
            ),
        ]
    )
    robot_description = {'robot_description': robot_description_content}
    robot_controllers = PathJoinSubstitution(
        [
            FindPackageShare('gz_ros2_control_demos'),
            'config',
            'diff_drive_controller_velocity.yaml',
        ]
    )

    node_robot_state_publisher = Node(
        package='robot_state_publisher',
        executable='robot_state_publisher',
        output='screen',
        parameters=[robot_description]
    )

    gz_spawn_entity = Node(
        package='ros_gz_sim',
        executable='create',
        output='screen',
        arguments=['-topic', 'robot_description', '-name',
                   'diff_drive', '-allow_renaming', 'true'],
    )

    joint_state_broadcaster_spawner = Node(
        package='controller_manager',
        executable='spawner',
        arguments=['joint_state_broadcaster'],
    )
    diff_drive_base_controller_spawner = Node(
        package='controller_manager',
        executable='spawner',
        arguments=[
            'diff_drive_base_controller',
            '--param-file',
            robot_controllers,
            ],
    )

    return LaunchDescription([
        # Launch gazebo environment
        IncludeLaunchDescription(
            PythonLaunchDescriptionSource(
                [PathJoinSubstitution([FindPackageShare('ros_gz_sim'),
                                   'launch',
                                   'gz_server.launch.py'])]),
            launch_arguments=[('world_sdf_file', 'empty.sdf'),]),
        RegisterEventHandler(
            event_handler=OnProcessExit(
                target_action=gz_spawn_entity,
                on_exit=[joint_state_broadcaster_spawner],
            )
        ),
        RegisterEventHandler(
            event_handler=OnProcessExit(
                target_action=joint_state_broadcaster_spawner,
                on_exit=[diff_drive_base_controller_spawner],
            )
        ),
        node_robot_state_publisher,
        gz_spawn_entity,
        # Launch Arguments
        DeclareLaunchArgument(
            'use_sim_time',
            default_value=use_sim_time,
            description='If true, use simulated clock'),
    ])
```

Amronos, 2024-11-17:
I tested this with a binary installation of v1.0.7 from the testing repository. The result is the same: using `gz_sim.launch.py`, the plugin is found successfully while when using `gz_server.launch.py`, the plugin is not found.

Tacha-S, 2024-11-26:
I encountered the same issue. Assuming it was an environment variable problem, I added the following, and it seems the loading is working correctly.

https://github.com/gazebosim/ros_gz/blob/5c1251a658ad8dd276664c2c88300d27572e25da/ros_gz_sim/launch/gz_sim.launch.py.in#L43-L96

However, there is still another problem.  
When executed as a component, the controller manager is also started in the same container, causing the controller manager's name to be identical to the container name. This leads to the following issues:

- The namespace specified in files like `ros2_control.xacro` is ignored.
- Services like `controller_manager/list_controllers` are prefixed with `container_name/***`.

Amronos, 2024-11-29:
> I encountered the same issue. Assuming it was an environment variable problem, I added the following, and it seems the loading is working correctly.

@Tacha-S could you make a PR with those changes?

PR #659 (2024-12-05): Set env path

# 🦟 Bug fix

Fixes #630 

## Summary

I set the environment variables to correctly load plugins when launching with a composable node.

## Checklist
- [ ] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck` passed (See [contributing](https://gazebosim.org/docs/all/contributing#contributing-code))
- [ ] All tests passed (See [test coverage](https://gazebosim.org/docs/all/contributing#test-coverage))
- [ ] While waiting for a [... cut]
PR #661 (2024-12-06): Set env path (backport #659)

# 🦟 Bug fix

Fixes #630 

## Summary

I set the environment variables to correctly load plugins when launching with a composable node.

## Checklist
- [ ] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck` passed (See [contributing](https://gazebosim.org/docs/all/contributing#contributing-code))
- [ ] All tests passed (See [test coverage](https://gazebosim.org/docs/all/contributing#test-coverage))
- [ ] Whi [... cut]
closed 2024-12-05 by ahcorde