gazebosim/ros_gz #657: Update binary release on humble repository

therealjohnpitt, 2024-11-27:
## Environment
* OS Version: Ubuntu 22.04 / ROS2 Humble

## Description
* Expected behavior: Expected version 0.244.16 in the repositories
* Actual behavior: Version 0.244.15 is the last one available

## Steps to reproduce

1. Run a humble docker container : 

```bash 
docker run -it ros:humble
```

3. List the available gazebo binaries: 

```bash 
apt update && apt search ros-humble-ros-gz
```

This would add EntityWrench support without the need to build from source as v0.244.16 has been released since July 2024 so a while ago.

Thanks !

ahcorde, 2024-11-27:
The release was done 4 months ago https://github.com/gazebosim/ros_gz/commits/humble/ . I will check if something went wrong

ahcorde, 2024-11-29:
I made the tag https://github.com/gazebosim/ros_gz/tree/humble but not the release https://github.com/ros/rosdistro/blob/master/humble/distribution.yaml#L8363C1-L8363C26

This PR should fix it https://github.com/ros/rosdistro/pull/43731

closed 2024-11-29 by ahcorde

therealjohnpitt, 2024-11-29:
Thanks, much appreciated !