gazebosim/ros_gz #682: Spawning heightmap model from nav2_map_server occupancy grid

damanikjosh, 2025-01-09:
## Desired behavior
I have been wanted a simpler approach to spawn 3D model from occupancy grid built using SLAM. I need it to do quick test of the navigation system using the environment that match the real scale.

## Alternative approach
The idea is to take map either from listening to `/map` topic or requesting `/map_server/map` service, preprocess and download it as image and spawn the model consisting  using UserCommand service `/world/<world_name>/create`

## Implementation suggestion
I have implemented a ROS package in my personal repository, but have no idea where I can contribute to this.
https://github.com/damanikjosh/heightmap_spawner

azeey, 2025-01-09:
That looks like a great project! 

We're hoping to implement a way for projects like yours to be easily discoverable and usable in the Gazebo ecosystem without having to contribute the plugin upstream (I.e in gz-sim or ros-gz). The issue is being tracked by https://github.com/gazebosim/gz-sim/issues/1995 and we hope to implement it this year. In the mean time, I've proposed that we add a page in the gazebosim.org/docs site to list projects that use Gazebo https://github.com/gazebosim/docs/pull/561. If that gets merged, I suggest opening a PR there to add your project.

kscottz, 2025-01-09:
@damanikjosh this is a fantastic project. It may take us a while to figure out how to get it to land. In the meantime we would like to share it with the broader ROS community. It might help if we found some time to chat. Can you reach out to me at katscott@intrinsic.ai?

closed 2025-01-14 by azeey