gazebosim/ros_gz #685: Problem with message transfer from Gazebo to ROS

wzhugit, 2025-01-20:
## Environment
OS Version: Ubuntu 22.04.5 LTS
* Source or binary build?
  
  
Source
branch:ros2

## Description
* Expected behavior: 
When publishing messages from Gazebo via 'gz topic --topic /chatter -m gz.msgs.StringMsg -p 'data:"Hello"'
 in one terminal and 'ros2 topic echo /chatter' in another while 'ros2 run ros_gz_bridge parameter_bridge /chatter@std_msgs/msg/String@gz.msgs.StringMsg' on a background terminal C, there should be a message printed to the  'ros2 topic echo /chatter' terminal.
* Actual behavior: 
On terminal C which running 'ros2 topic echo /chatter', there is nothing printed when  'ros2 run ros_gz_bridge parameter_bridge /chatter@std_msgs/msg/String@gz.msgs.StringMsg'  terminal on background.
## Steps to reproduce

1.Create a workspace "mkdir bridge_ws" and create src directory in it "cd bridge_ws;mkdir src"
2.run "rosdep install -r --from-paths src -i -y --rosdistro humble" and "#All required rosdeps installed successfully" is returned.
3.run "source /opt/ros/humble/setup.bash"
4.run "colcon build" and wait
5.open a new terminal, run "cd bridge_ws" ,"source /opt/ros/humble/setup.bash" and "source install/setup.bash",then run "ros2 run ros_gz_bridge parameter_bridge /chatter@std_msgs/msg/String@gz.msgs.StringMsg",the terminal should return something similar to “[INFO] [1737355395.023715821] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/chatter (gz.msgs.StringMsg) -> /chatter (std_msgs/msg/String)] (Lazy 0)
[INFO] [1737355395.026791266] [ros_gz_bridge]: Creating ROS->GZ Bridge: [/chatter (std_msgs/msg/String) -> /chatter (gz.msgs.StringMsg)] (Lazy 0)
[INFO] [1737355649.022489895] [ros_gz_bridge]: Passing message from ROS std_msgs/msg/String to Gazebo gz.msgs.StringMsg (showing msg only once per type)
”
6.Open a new terminal and run "source /opt/ros/humble/setup.bash" "ros2 topic echo /chatter"
7.Open a new terminal and run "gz topic --topic /chatter -m gz.msgs.StringMsg -p 'data:"Message from Gazebo"'" and "Message from Gazebo" should be printed to terminal which running "ros2 topic echo /chatter"

## Output

Nothing appears in the terminal which running "ros2 topic echo /chatter"

[image]

azeey, 2025-01-31:
Can you please provide what version of Gazebo you're using? If you have multiple versions of Gazebo (e.g. Harmonic and Fortress) installed, you might have compatibility issues.

wzhugit, 2025-02-02:
The Gazebo Version is Gazebo Garden v7.9 and the ROS version is ROS2 Humble. There is only one Gazebo and one ROS installed.

azeey, 2026-03-11:
I'll go ahead and close this since there hasn't been any activity here. Please feel free to reopen if you still have the issue.

closed 2026-03-11 by azeey