gazebosim/ros_gz #687: SpawnEntity.srv support??

tanmay2798, 2025-01-22:
I want to ask if SpawnEntity.srv service supported in ros2 humble branch? And if so, what should be the command?

I found the command for ControlWorld - ros2 service call /world/<world_name>/control ros_gz_interfaces/srv/ControlWorld "{world_control: {pause: false}}"

But I cannot run SpawnEntity.srv as it always says waiting for service and nothing happens.

Even running ros2 run ros_gz_bridge parameter_bridge /world/shapes/control@ros_gz_interfaces/srv/ControlWorld command for SpawnEntity starts no service.

rafiqrana, 2025-03-19:
Was it solved?

masterccioli, 2025-04-29:
> Was it solved?

Yes: https://github.com/ros-simulation/simulation_interfaces/pull/1